itemisCREATE/statecharts #1364: Provide support for xtext based rename refactorings

jdicks, 2017-05-09:
This will have 2 aspects: 
1. update cross-references within specifications based on a change outside the SCT resource
2. update all cross-references triggered from within an embedded resource context in SCT

The first aspect can be handled by implementing an IReferenceUpdater for the SCT resource.
This reference updater have to apply text changes calculated by xtext to specification features properly.

The second aspect is accessible through the SCT embedded editor > right click > Refactoring > 'Rename..'.
The current implementation is based on Stext but could be based on Xtext (RenameelementProcessor) so that it can be applied for all Xtext based languages integrated via the sct domains embedded editor feature. The element rename processor will delegate to all IReferenceUpdater contributions for resources where cross references can be found. So a refactoring of an embedded element will use the IReferenceUpdater to update the specifications placed logically in the same file as the elements of the embedded fake resource.

jdicks, 2017-05-11:
Following things left:
* renaming of e.g. variable definitions fail
  * owner of the definition is transient
  * usage of variables is already working
* calculate element to rename from selection
  * current implementation fails if the rename refactoring is triggered from e.g. a variable definition itself
* tests for 
  * SCTXtextSpecificationReferenceUpdater
    * used propagate updates for specifications within an sct resource
  * RenameEmbeddedXtextReferenceHandler
    * hard to test without some kind of ui test framework
    * have to be discussed

Other issues: 
* presentation of changes in preview isn't optimal

jdicks, 2017-08-08:
i rebased the current state and pushed it to Yakindu/statecharts/tree/issue_1364_integration

andreasmuelder, 2017-09-12:
scheduling this for backlog as discussed

svenjawendler, 2017-09-13:
Rename Refactoring currently does not work on event names:
[image]