itemisCREATE/statecharts #1379: NPE in Eclipse Log

paphko, 2017-05-18:
I found (repeating) NPE in Eclipse log:
```
java.lang.NullPointerException
	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.createExecutor(GeneratorExecutorLookup.java:66)
	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.execute(GeneratorExecutorLookup.java:50)
	at org.yakindu.sct.generator.builder.EclipseContextGeneratorExecutorLookup$1.run(EclipseContextGeneratorExecutorLookup.java:43)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
```
and
```
java.lang.NullPointerException
	at org.yakindu.sct.model.sexec.naming.DefaultNamingService.asIdentifier(DefaultNamingService.java:468)
	at org.yakindu.sct.generator.c.CNamingService.asEscapedIdentifier(CNamingService.java:255)
	at org.yakindu.sct.generator.c.Naming._access(Naming.java:524)
	at org.yakindu.sct.generator.c.Naming.access(Naming.java:674)
	at com.yakindu.sct.domain.c.runtime.generator.CExpressionCode._code(CExpressionCode.java:135)
	at com.yakindu.sct.domain.c.runtime.generator.CExpressionCode.code(CExpressionCode.java:270)
	at org.yakindu.sct.generator.c.ExpressionCode._code(ExpressionCode.java:95)
	at org.yakindu.sct.generator.c.ExpressionCode.code(ExpressionCode.java:530)
	at org.yakindu.sct.generator.c.ExpressionCode._code(ExpressionCode.java:355)
	at org.yakindu.sct.generator.c.ExpressionCode.code(ExpressionCode.java:536)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:361)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:544)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.FlowCode._code(FlowCode.java:394)
	at org.yakindu.sct.generator.c.FlowCode.code(FlowCode.java:556)
	at org.yakindu.sct.generator.c.StatemachineSource.enterFunction(StatemachineSource.java:326)
	at org.yakindu.sct.generator.c.StatemachineSource.content(StatemachineSource.java:167)
	at org.yakindu.sct.generator.c.CGenerator.generateArtifacts(CGenerator.java:89)
	at org.yakindu.sct.generator.c.CGenerator.generate(CGenerator.java:80)
	at org.yakindu.sct.generator.core.execution.SExecGeneratorEntryExecutor.execute(SExecGeneratorEntryExecutor.java:54)
	at org.yakindu.sct.generator.core.execution.AbstractGeneratorEntryExecutor.execute(AbstractGeneratorEntryExecutor.java:58)
	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.execute(GeneratorExecutorLookup.java:51)
	at org.yakindu.sct.generator.builder.EclipseContextGeneratorExecutorLookup$1.run(EclipseContextGeneratorExecutorLookup.java:43)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:55)
```

terfloth, 2017-07-20:
@paphko : what model caused the exceptions?

andreasmuelder, 2017-07-21:
This exceptions can only be thrown when the model is invalid and the build automatically feature is disabled. The code generator tries to generate an invalid model then that results in the NPE Exception.

paphko, 2017-07-22:
Then I would suggest to wrap code that deals with user input (such as possibly invalid models) into try-catch blocks and to handle such unexpected errors with a proper error message.

andreasmuelder, 2017-07-27:
Or we force model validation before we generate the code.

andreasmuelder, 2017-07-27:
The first exception is fixed, for the second one i need a model to reproduce this.

andreasmuelder, 2017-07-28:
@paphko please file a new issue with a test model if this exception still exists.

closed 2017-07-28 by andreasmuelder

paphko, 2017-07-28:
Thank you @andreasmuelder!
Unfortunately, I don't have the errornous model available.