itemisCREATE/statecharts #1383: Select named interface as standard in regions with same name

BeckmaR, 2017-05-19:
I just had an idea that might be helpful to keep qualifiers in the statechart short, but I don't know how complicated it might be.

Lets say I have a named interface:
```
interface Numpad:
  in event key: integer
```
Normally and right now I would have to access the event with `Numpad.key` and `valueof(Numpad.key)`. That gets long and annoying quickly.

I think it would be quite cool to auto-select this interface as the default interface with top priority if I the top-most region I am in is named the same as the interface.

So, in all states that are directly contained in a region named Numpad, I could just write `key`. The more complicated part would be stuff that exists in both the default and a named interface. For example:
```
interface:
  in event key: integer
interface Numpad:
  in event key: integer
```
Maybe I could then access it with `default.key`?

andreasmuelder, 2017-07-04:
I think that  this is not the way stastecharts are modeled. You usually do not have a region per interface. You access elements from different interfaces from one region.

BeckmaR, 2017-07-04:
That might be true.

closed 2017-11-10 by BeckmaR