itemisCREATE/statecharts #1386: Synchronization fires even when not all incoming states are active

BeckmaR, 2017-05-19:
Synchronizations should only fire their outgoing transitions when source states are active.

However, a composite state containing an exit state can break this behavior at least in the simulation and can exit fire the synchronization without the other source state being active.

Example project attached. Just set x to 4 in the simulation, and exit becomes active without a being active.
[Synchronization.zip](https://github.com/Yakindu/statecharts/files/1014427/Synchronization.zip)

BeckmaR, 2017-05-19:
[image]

terfloth, 2017-07-20:
nice test case ...

Assumption:
The transition from 'State1' to synch is also interpreted as the transition that has to be taken in the case of an exit. In this scenario an additional transition that does not target a sync node is required.

We need to:

- update the sequencer
- add validation

tkutz, 2018-07-12:
I need some more clarification here. Let's take this slightly adjusted example. What shall happen when `x=1` is set. Shall transition `Composite_A->F` fire?

[image]

PR #2137 (2018-07-13): Issue 1386 exit on sync problem
Fixes #1386
closed 2018-07-13 by terfloth