itemisCREATE/statecharts #1392: [GUI] Notes edition is not practical

wixiw, 2017-05-24:
If a some difficulties to edit notes efficiently in the state machine graphical diagram editor. When I double click on a note the field is not editable, since I would expect it enters edition mode.

At present time I often go quickly in copying current text, deleting the note and creating a new one, which is not the expected workflow.

BeckmaR, 2017-05-24:
I have to agree. I experienced the same problems with nootes and would say they are 90% broken right now.

andreasmuelder, 2017-06-12:
This is an usability issue, the notes still use the old style to enter text edit mode (single click the label and wait) while all others use the new style (double click.)

PR #1421 (2017-06-13): Adds double click drag tracker to notes
resolves #1392
commit 50a84ac5 (2017-06-13): Adds double click drag tracker to notes 

resolves #1392
closed 2017-06-13 by BeckmaR