itemisCREATE/statecharts #1394: [GUI] action parser refuse lines finishing with ";"

wixiw, 2017-05-24:
Currently, when I finish a line with a ";" after an action description, the parser generates an error. I would expect the parser to accept the line and consider the ";" as useless but ok. 

Current behavior is annoying because I often add/remove actions during development or use copy/paste to share information between states/transitions. Having to modify the line is painfull and useless. 

I would like to be able to do the following in a state corpus:
```
entry/
action1();
action2();

exit/
action3();
```

BeckmaR, 2017-05-24:
Would agree with this. I stumble upon this very often as well and it seems arbitrary. Shouldn't we be able to change this without breaking current statecharts @terfloth?

BeckmaR, 2017-06-13:
Duplicate of #1047, closing.

closed 2017-06-13 by BeckmaR