itemisCREATE/statecharts #1441: java.lang.IllegalStateException: No dispatch function

cynergizer, 2017-06-20:
I imported a project into the pro version, which built fine in the free version. When I try to generate code from the PRO version, I get this error message:
java.lang.IllegalStateException: No dispatch function

BeckmaR, 2017-06-20:
Hey @cynergizer, thank you for the report!

Can you share some more information with us, e.g. the version of Yakindu SCT you use (and used before), and if you actually just imported the project or did you change the statechart into a C-domain statechart? Also, can you provide a stacktrace? That would be extremely helpful for us.

andreasmuelder, 2017-07-04:
@BeckmaR  This seems to be the same problem as reported for the pro edition when using conditional expressions in C/  C++ Code Generator. I think this is fixed, plese recheck and close this issue.

BeckmaR, 2017-07-04:
That problem only exists when using C-Domain statecharts. It's not clear yet what @cynergizer actually did.

andreasmuelder, 2017-07-10:
closing this as fixed, please reopen if the error still exists in the upcoming version.

closed 2017-07-10 by andreasmuelder