itemisCREATE/statecharts #1442: Define new statechart specification section

BeckmaR, 2017-06-21:
## Problem
Several issues regarding the statechart specification section came up, also it was often mentioned in surveys and conversations. 

* #1347: Impossible to scroll in the interface widget, bad with huge interfaces
* #1278: Usage of the definition section as an editable text is not intuitive
* #1156: Auto-formatting of the spec section destroys user formatting
* #661: Rename refactoring doesn't work on Linux (spec section loses focus on right click)
* (no issue yet, please update) Statechart spec is not reusable across statecharts

tl;dr: It seems like the interface widget is unintuitive and somewhat awkward to use

## Solutions
### Multipage editor
It was proposed to use a multipage editor instead:
* #849: Split sc editor and interface widget into multipage editor
But another request was to improve the reusability of the interface section and it remains unclear how this solution might help that.
## New proposal: `*.stext` files
I want to propose another solution that seems intuitive and easy to use to me.
1.  Keep the interface widget section exactly the way it is
1. Introduce a new filetype: `*.stext` 
    * Can contain everything the spec section can (except maybe local reactions?)
1. Use the `import:` statement and allow to import `*.stext` ressources.
    * Maybe even allow `*.stext` files to import other ones again. 
This would allow the user to reuse events and interfaces across statecharts, but keep small interfaces right next to the statechart. Might be ideal for both usecases, small and big projects. And it would allow us to provide a proper editor that supports syntax highlighting.
### Contra
The downside would obviously be that older versions of Yakindu would not support the new import statement. One would need to decide if we have to bump the major version number then.

andreasmuelder, 2017-06-29:
There was another proposal regarding the interface section in #1151 what seems to be a good and compatible idea.

SimonEit, 2017-09-08:
While strongly supporting all mentioned solutions, a low level / low cost improvement to this issue would be that the statechart specification section does not lose the code highlighting when out of scope. 
I would also suggest the usage of a monospace font to make the appearance more "code-like". 

While obviously not solving the core problem changing these visuals should at least help new user to more easily distinguish the specification section as a codeable textarea.

andreasmuelder, 2018-01-02:
closing as  #1151 is implemented

closed 2018-01-02 by andreasmuelder