itemisCREATE/statecharts #1445: After creating a subdiagram, regions are positioned above each other

RainerKlute, 2017-06-21:
- Unpack [twoRegions.zip](https://github.com/Yakindu/statecharts/files/1092168/twoRegions.zip) and open the statechart:
[image]
- Right-click on the composite state C to open the context menu.
- Select _Refactor → Extract Subdiagram_.
- Open the subdiagram by clicking on the small symbol in state C.
Result: Only one of the regions is visible, because it covers the other. Move it to the side to unhide the other statechart.
[image]

svenjawendler, 2017-09-11:
Here is an animated gif demonstrating this problem!
[image]

andreasmuelder, 2018-02-05:
This will be fixed via https://github.com/Yakindu/statecharts/pull/1890

closed 2018-03-01 by andreasmuelder