itemisCREATE/statecharts #1447: Functional and usability analysis of diagram export

terfloth, 2017-06-23:
We provide basic diagram export functionality. Some features like Pdf export are not working properly and are platform specific.

The goal of this task is 
   1. analyze the status from a functional point of view on different platform.
   1. analyze the current status from a usability point of view

SimonEit, 2017-09-08:
I just tried to usability-test myself with the following goals: 
- Export a statechart diagram to PDF
- Export a statechart diagram to PNG/JPG/GIF (Image)
- Print a statechart on my printer
... without usage of Screenshots.

I am using V3.0.1 under Mac OS and failed miserably.

**1)** The export wizard is not exactly helpful.
[image]
None of these entries seem to bring me any closer to one of my goals

**2)** Printing from SCT seems to be broken:
[image]
I have several valid printers installed :)

svenjawendler, 2017-12-13:
This also occured in test 3.2.0. 
Enhancement: Could we delete JPEG and BMP from the list?