itemisCREATE/statecharts #1454: Spelling: "file name" should be changed to "filename"

RainerKlute, 2017-06-26:
In the UI, all occurences of "file name" should be changed to "filename".

Both spellings are correct, however, we should use one of them consistent in documentation and UI. The spelling "filename" seems to be more common, thus it should be the one of choice.

In the UI, there is at least the label "File name:" in the _New YAKINDU Statechart_ wizard, but there's probably more.

jdicks, 2017-06-28:
For 'File > New > File' "File name:" is used, for 'Mwe2 file' also... so use 'File name:' because we cannot change them.

RainerKlute, 2017-06-28:
We should change it wherever we can change it. (I didn't tell you yet that we have to change "modeling" to "modelling" due to the fact that we have to use British English now, did I? :-))

jdicks, 2017-06-28:
i meant: we cannot change it in every eclipse component we are using... so whenever someone thinks about changing... use 'File name:'  👍

RainerKlute, 2017-06-28:
Sure. That's what I meant with "wherever **we** can change it".

BeckmaR, 2017-06-28:
Nah. When eclipse decided to use _File name_ and there are x locations where that is used and we cannot change it and zero locations where _filename_ is used we should **not** use _filename_. Sometimes continuity > correctness.

andreasmuelder, 2017-06-28:
We will leave it as file name to be consitent to the eclipse ui.

closed 2017-06-28 by andreasmuelder