itemisCREATE/statecharts #1472: Hook into "Build Project"

BeckmaR, 2017-07-04:
_If_ this is possible at all, it would be a nice feature for C/C++ statecharts to hook into Eclipse / CDTs "Build project". At the time being, in the worst case, one needs to generate the statechart(s), the test(s), and then build the project. It's only a few clicks but it actually adds up.
It would be so much nicer if SCT would just listen to "Build Project", execute all sgens and then let Eclipse compile the C-Code.

andreasmuelder, 2017-07-04:
This should work. The builder builds all sgen files. Maybe our Builder is executed after the CDT Builder and this is the problem. Can you check this?

BeckmaR, 2017-07-04:
Nope, using CDT's "Build project" doesn't generate anything at all, neither before nor after the compilation.

## Example project
[covfefe.zip](https://github.com/Yakindu/statecharts/files/1121931/covfefe.zip)
1. Import project
1. Build project, should work as expected
1. Delete "src-gen"
1. Build project again, nothing gets generated. Sad.

BeckmaR, 2019-01-22:
So I dug around and found two / three issues:
* The initial project configuration puts the CDT build before the SCT build, so this will never work.
* During build, we don't (and currently can't) check if the generated files exist. If the sgen and the statechart didn't change since the last build, we don't build again in an incremental build.
* Even if the build does happen and happens in the correct order, there is still a problem - we *schedule* the build job instead of executing it directly on the current `ProgressMonitor`. This means that the current build will always execute fully before our build job is run, because scheduling puts it at the end of the queue. Instead we should just generate if we are called from the `BuildManager`.

This is all of course not an issue in smaller projects where "_Build automatically_" is active and we generate on every save. But bigger projects where the user deactivates this and instead builds on demand, or if the statechart was changed via git and not manually regenerated, we might come too late in the process.