itemisCREATE/statecharts #1503: Tracking a simulation impossible when launched with only subdiagram open

RainerKlute, 2017-07-14:
1. Create a statechart model with a subdiagram.
1. Open the subdiagram.
1. Close the parent tab.
1. Start the simulation.
1. Since you closed the parent tab, you cannot see where the simulation is, i.e. which state is active.
1. Opening the statechart from the model file does not help, since all states are displayed as inactive.

Workaround: Don't close the parent tab.

andreasmuelder, 2017-08-02:
I can not reproduce this. When a subdiagram is open, the active states get highlighted as long as the state that contains the subdiagram is active of course. 
After opening the statechart from the model file, it is neccessary to select the running simulation session from the Debug View.
Maybe this is a little bit inconvenient from a usability perspective.
If only one session is running for a certain model we could connect this debug session to it.

BeckmaR, 2017-09-04:
So basically what should be done is:

If a user opens a statechart model in the Debug view and a debug session is already running, the debug view should open said session automatically.

...and if two sessions are running? Connect to the first one? Maybe forbid running two sessions at once completely? I find myself having multiple sessions open unintentionally often.

closed 2017-11-28 by andreasmuelder

robert-rudi, 2017-11-28:
When the user closes the root diagram while in a subdiagram, the user will be prompted if all subdiagrams should also be closed. If the users hits "No" and the subdiagram remains open, a simulation can be triggered anyway. In this case, the root diagram will be opened again programmatically for the context of the current statechart. The tracking of the simulation is now possible, because the root diagram will be reopened and the active states are highlighted.

commit 24a36687 (2017-11-28): Fixes #1503 If the corresponding parent diagram is closed when starting a simulation from one of its subdaigrams, the root diagram will now be opened so the currently active state is visible. (#1758)