itemisCREATE/statecharts #1519: Downcast of scope object in custom code generator not working.

PGaus, 2017-07-19:
In an custom Xtend generator we want to use following code:
`switch (scope) {
	InterfaceScope: scope.gen
	default: scope.toString
}
`
But the case is never reached. With each scope always the default is called.

With the two projects in [workspace.zip](https://github.com/Yakindu/statecharts/files/1159088/workspace.zip) we have an example showing this problem because there you see that line 33 of TestGenerator.xtend is never reached.  
Line 34 is the evidence that the scope object is of the type `org.yakindu.sct.model.stext.stext.impl.InterfaceScopeImpl`

After investigation into the problem a possible root cause could be that there are two different classloaders involved. You can see these if you execute `return scope.class.classLoader.toString + " " + InterfaceScopeImpl.classLoader.toString`

We already had a look to org.yakindu.sct.generator.runner.WorkspaceGeneratorExecutor.createInstance(GeneratorEntry, String) where the classLoader for generator runner is instantiated.

andreasmuelder, 2017-07-25:
I can reproduce this on my machine. As suggested, it seems to be a class loading issue that has to be investigated further.

andreasmuelder, 2018-09-18:
As discussed, we will remove the feature to develop code generators in the runtime in the next version 3.5.0. To develop a code generator, a developer workspace is required. Basic know-how about plugin development (PDE) is required.

PR #2502 (2018-12-07): resolves classloading issues for stext elements
fixes #1519 

With the dependency to SText the equinox class loader can reuse the already loaded Stext classes
commit 062eeb8d (2018-12-07): resolves classloading issues for stext elements (#2502)

fixes #1519 

With the dependency to SText the equinox class loader can reuse the already loaded Stext classes
closed 2018-12-07 by andreasmuelder