itemisCREATE/statecharts #1530: Welcome page: links to forum and issue tracker don't work

RainerKlute, 2017-07-24:


tkutz, 2017-07-24:
Works for me under MacOSX

RainerKlute, 2017-07-24:
Okay, I assigned it to myself to investigate further on Linux.

BeckmaR, 2017-07-25:
Works on my Linux machine. Ubuntu 16.10. Make sure you have a browser configured in SCT (Preferences - type "Browser" in the search field)

RainerKlute, 2017-09-12:
Retested with the most recent product build and it still doesn't work for me in the default configuration.

I noticed the following messages on the console when clicking on the links:
```
env: „/some/dir/yakindu-sct/https:/groups.google.com/forum/#!forum/yakindu-user“: Datei oder Verzeichnis nicht gefunden
env: „/some/dir/yakindu-sct/https:/github.com/yakindu/statecharts/issues“: Datei oder Verzeichnis nicht gefunden
```

After changing the external browser in the preferences from "Default system web browser" to "Firefox", it works as expected. However, the user should not be required to do so in order to be able to access forum and issue tracker.

andreasmuelder, 2018-01-04:
Platform dependend, closing

closed 2018-01-04 by andreasmuelder