itemisCREATE/statecharts #1531: Wrong letter cases in error message

RainerKlute, 2017-07-24:
The error message

```Outgoing Transitions from Entries can not have a Trigger or Guard.```

should read instead:

```Outgoing transitions from entry points can not have a trigger or guard.```

rherrmannr, 2017-07-25:
wrong latter case, too

`Outgoing Transitions from Entries can only target to sibling or inner states.`

RainerKlute, 2017-07-25:
There are likely many more of those. 😭

BeckmaR, 2017-07-26:
That's a dup of the issue #559

closed 2017-07-26 by BeckmaR

BeckmaR, 2017-07-26:
...and spelling errors are not bugs

RainerKlute, 2017-07-26:
Oh, of course they are! Spelling errors are creating images in people's minds on how serious we are taking matters. Users might draw their own conclusions, conveying their impression on spelling quality to that of code quality, like it or not.