itemisCREATE/statecharts #1533: [Linux] Generator project wizard doesn't validate text pasted via middle mouse button

RainerKlute, 2017-07-24:
If on Linux you select the generator class name using the left mouse botton and then paste it into the generator project wizard's *Generator Class* field using the middle mouse button – as opposed to [Ctrl+C], [Ctrl+V] – the text seems not to be properly recognized, validated, whatever.

The effect is that the *Finish* button is not activated even though the input fields contain valid data.

Checking and unchecking *Configure for Plugin Export* helps. However, this workaround is less than intuitive for most users.

andreasmuelder, 2017-07-26:
This is an SWT Bug, please file an external bug in eclipse bugzilla.

RainerKlute, 2017-07-26:
Hm, I am not so sure that this is an SWT bug. For a comparison, I tried the Java project wizard with middle mouse pasting, and it works as expected.

andreasmuelder, 2018-01-02:
ok, maybe @BeckmaR can have a look at this as I am not a Linux user.

rherrmannr, 2018-01-11:
This bug also occurs if you paste sth into the ``Project name:`` field

rherrmannr, 2018-09-04:
This is already fixed with https://github.com/Yakindu/statecharts/pull/1856

closed 2018-09-04 by rherrmannr