itemisCREATE/statecharts #1561: [Documentation] Write a new section on execution semantics

RainerKlute, 2017-07-28:


andreasmuelder, 2018-01-10:
@Rainer so this can be closed?

RainerKlute, 2018-01-10:
Reopened this issue since this is still not done. We don't describe the execution semantics properly or at least concisely and completely yet.

RainerKlute, 2018-08-10:
The work I have done so far is on the _doc/1561_execution_semantics_ branch. I covered the following topics:
* The state vector
* Initializing a state machine
* Entering and exiting a state machine
* Entering and exiting a region
* Entering and exiting a state
* The run cycle
  * React methods of states
* Checking the active state for actions to be performed
* Transitioning from one state to another state

However, the explanations are still incomplete and have a varying level of details. They should not yet be merged into the master.

closed 2020-11-26 by terfloth