itemisCREATE/statecharts #1594: Compile error in CPP code for string assignments within conditional expressions

tkutz, 2017-08-04:
To reproduce, take digital_watch example from example repository and generate CPP code from it.

You will find some errors, one of which is this one:

[image]

Normal assignments to string variables work fine. 
It seems like CPP behaves different regarding type conversion when the string is inside another expression, here a conditional expression.

rherrmannr, 2017-10-25:
I can reproduce this error. 
You cant write a string into the text, which is sc_string char*.

A solution may be using strcpy:
```
void Watch::effect_modes_Regular__region0_Alarm_2_lr1_lr1()
{
	ifaceDisplay.first = (ifaceInternalSCI.alarm_2_time / 36000) % 24;
	ifaceDisplay.second = (ifaceInternalSCI.alarm_2_time / 600) % 60;
	ifaceDisplay.third = (ifaceInternalSCI.alarm_2_time / 10) % 60;
	//ifaceDisplay.text = (ifaceInternalSCI.alarm_2_enabled) ? "Alarm 2 ON" : "Alarm 2 OFF";
	if (ifaceInternalSCI.alarm_2_enabled){
		strcpy(ifaceDisplay.text, "Alarm 2 ON");
	}else {
		strcpy(ifaceDisplay.text, "Alarm 2 OFF");
	}
}
```
I will figure out if this will work.

rherrmannr, 2017-10-26:
Using (char*) conversion instead
```ifaceDisplay.text = (ifaceInternalSCI.alarm_1_enabled) ? (char*)"Alarm 1 ON" : (char*)"Alarm 1 OFF";```

PR #1731 (2017-10-26): Compile error in CPP code for string assignments within conditional expressions
fix #1594
closed 2017-10-26 by tkutz