itemisCREATE/statecharts #1595: No escaping of ambiguous names for CPP generator

tkutz, 2017-08-04:
To reproduce, use digital_watch example from examples repository and generate CPP code from watch_subdiagrams.sct into a CPP project.

The model has variables "time" and "set_time" which results in ambiguous elements in the generated CPP code.

rherrmannr, 2017-10-26:
for a simple example define following variables in a statechart
```
var test:integer = 0
var set_test : integer = 0
var get_test : integer = 0
```

rherrmannr, 2017-10-26:
A solution proposal: Prohibit variable names, which begins with ``set_`` and ``get_``

@andreasmuelder  are there already any implementations for the init function or something else?

rherrmannr, 2018-01-11:
Can be closed after discussion with @BeckmaR and @andreasmuelder 

``shit in, shit out``

closed 2018-01-11 by rherrmannr