itemisCREATE/statecharts #1634: Effective C++ comliant code - Three warnings occur by using timer

rherrmannr, 2017-09-05:
Fix compiler warning if you are using -Weffc++ compiler flag:

```
warning: ... has pointer data members [-Weffc++]
warning:   but does not override ... [-Weffc++]
warning:   or 'operator=(const ...)' [-Weffc++]
```

This error occurs, because the timer gets declared as a pointer:
`` TimerInterface* timer; `` 

Therefore, a copy constructor and an assignment operator must be declared.

rherrmannr, 2017-09-05:
Easiest fix by using Item 6:
Declare copy constructor and assignment operator private. 

see "Effective C++" 
``Item 6: Explicitly disallow the use of compilergenerated
functions you do not want.`` 

This works, but then it's no more possoble to assign or copy statechart objects:
```
Default d;
Default b;
b=d 
```
```
Default d;
Default b(d);
```

rherrmannr, 2017-09-05:
Using proposals of Item 11
``Item 11: Handle assignment to self in operator=.``

results an implementation like this:
```
Default& Default::operator=(const Default& rhs)
{
	TimerInterface *temp = timer;
	timer = new TimerInterface(*rhs.timer); // this line leads to compile errors
	delete temp;

	return *this;
}
```
Which leads to compile errors. Pure virtual methods of the timer interface must be implemented first.

rherrmannr, 2017-09-05:
Copy constructor could be implemented like this:

``Default::Default(const Default&rhs):timer(rhs.timer),stateConfVectorPosition(rhs.stateConfVectorPosition),iface(rhs.iface){ }`` 

My suggestion:
Declare assignment operator private and copy constructor public.

BeckmaR, 2017-09-08:
I don't really like the idea to implement methods that literally no one needs just to conform with some book. Wouldn't it maybe be easier to remove the pointer member?

rherrmannr, 2017-09-13:
@terfloth what's your opinion?

closed 2017-09-21 by andreasmuelder

elgeekomm, 2017-10-12:
Thanks, this is helpful! Our codebase is Weffc++ compliant and we had to disable this feature for our generated Statecharts, which is something we would like to avoid.