itemisCREATE/statecharts #1643: [Documentation] Variables in SGen

andreasmuelder, 2017-09-12:
Variables can now be used in SGen like this:

GeneratorModel for yakindu::java {

 var targetSubFolder : string = "mySubFolder"
 var generateTimerServce : boolean = true

statechart ECU_B {   

    feature Outlet {
        targetProject = "TargetProject"
        targetFolder = "target/" + targetSubFolder
        libraryTargetFolder = "src"
    }  
    
    feature GeneralFeatures {
        TimerService = generateTimerServce
    }
}
}

These variables can also be overriden for the headless build with scc -v name=value;name2=value2

BeckmaR, 2017-09-14:
Also, do not forget to update the output of `scc` in the documentation (or revalidate if it's actually needed there)

RainerKlute, 2017-09-22:
Incorporated in pull request #1676.

closed 2017-09-22 by RainerKlute