itemisCREATE/statecharts #1659: Java code generators: inconsistent targetFolder behavior

RainerKlute, 2017-09-19:
The Java code generators for state machines and for SCTUnit tests behave inconsistently regarding the Outlet feature's _targetFolder_ property if the generator model is located in a subdirectory, say _codegen_.

The state machine code generator interprets a relative _targetFolder_ value like "src-gen" as being relative to the directory the generator model is located in and creates the directory _codegen/src-gen_. A path relative to the project root can be specified as ```targetFolder = "/src-gen"```.

The SCTUnit code generator, however, interprets a relative _targetFolder_ value like "src-gen" as being relative to the project root and creates the directory _src-gen_.

**The SCTUnit code generator should behave like the state machine code generator.**

andreasmuelder, 2017-09-21:
@RainerKlute please move this to scpro repo

RainerKlute, 2017-09-22:
Moved to sctpro repository.

closed 2017-09-22 by RainerKlute