itemisCREATE/statecharts #1662: SCTUnit code generator pulls folder names into package names

RainerKlute, 2017-09-19:
I have the following generator model:
```
GeneratorModel for sctunit::java {

	test foo.light_switch.test.light_switch_tests {

		feature Outlet {
			targetProject = "light_switch"
			targetFolder = "src-gen/sctunit/java"
		}

		feature Naming {
			basePackage = "bar.light_switch.tests"
		}

		feature StatechartNaming {
			basePackage = "bar.light_switch"
			implementationSuffix = "impl"
			libraryTargetFolder = "src-gen/sctunit/java_ltf"
		}

	}
}
```

I'd expect the package name of the generated classes to be _bar.light_switch.tests.foo.light_switch.test_. However, it is _sctunit.java.bar.light_switch.tests.foo.light_switch.test_. Obviously the code generator pulls the components of the _targetFolder_ into the package name, except for the first one.

andreasmuelder, 2017-09-21:
@RainerKlute please move this to pro repo

RainerKlute, 2017-09-22:
Moved to sctpro repository.

closed 2017-09-22 by RainerKlute