itemisCREATE/statecharts #1679: C++ generator: inconsistent types used for index into stateConfVector (signed / unsigned compare)

DanielBernhard, 2017-09-25:
The maxOrthogonalStates variable (used to define the dimension of the stateConfVector array) is an sc_integer (e.g. int32_t) while the stateConfVectorPosition variable (used to access the array) is an sc_ushort (e.g. uint_fast16_t).

This is inconsistent and leads to "signed/unsigned compare" warnings of the compiler.

While I agree that the actual danger is non-existing (I don't assume that we ever have more than 16'000 orthogonal states) I don't think that this would be very hard to fix (change maxOrthogonalStates to sc_ushort and fix the type of the loop variable in the init function).
This would lead to consistent code (and fewer compiler warnings...)

terfloth, 2017-09-25:
I completely agree. Issue is scheduled ...

BeckmaR, 2017-09-27:
@terfloth - should we use `sc_integer` or `sc_ushort`for both?

andreasmuelder, 2018-01-05:
@BeckmaR I think we can use sc_ushort as proposed

terfloth, 2018-01-08:
Yes - sc_ushort fits well

andreasmuelder, 2018-01-08:
fixed via #1845

closed 2018-01-08 by andreasmuelder