itemisCREATE/statecharts #1705: Add namespace support for C++ code generator

terfloth, 2017-10-11:
The C++ code generator should provide a feature to specify the namespaces that should be used by:

- the generated state machine classes
- the interface types like StjatemachineInterface, TimedStatemachineInterface, TimerInterface...

tkutz, 2018-07-10:
We could use the namespace feature from the definition section.

BeckmaR, 2018-07-18:
@tkutz However, it would probably not be useful to put the `StatemachineInterface` and the likes into the namespace of the statemachine - consider the case where you have two statecharts with different namespaces, generate both into the same folder, the first statechart's interfaces would be overwritten by the second one.

PR #2289 (2018-09-26): Issue 1705 2
Resolve #1705: C++ generator now includes the namespace in the generated source files.
closed 2018-09-26 by BeckmaR