itemisCREATE/statecharts #1710: Help: typescript-domain-toc.xml and c-domain-toc.xml are missing

Martin-He, 2017-10-13:
See screenshot for installed plug-ins and features
[image]
[image]

The following text was written into the log:

!ENTRY org.eclipse.help 4 0 2017-10-13 11:40:20.093
!MESSAGE Error reading help table of contents file /"org.yakindu.sct.doc.user/help/user-guide/typescript-domain-toc.xml" (skipping file)
!STACK 0
java.io.FileNotFoundException
	at org.eclipse.help.internal.toc.TocFileParser.parse(TocFileParser.java:44)
	at org.eclipse.help.internal.toc.TocFileProvider.getTocContributions(TocFileProvider.java:45)
	at org.eclipse.help.internal.toc.TocManager.getAndCacheTocContributions(TocManager.java:211)
	at org.eclipse.help.internal.toc.TocManager.getTocContributionsForToc(TocManager.java:199)
	at org.eclipse.help.internal.toc.TocManager.getRootTocContributions(TocManager.java:307)
	at org.eclipse.help.internal.toc.TocManager.getTocs(TocManager.java:68)
	at org.eclipse.help.internal.webapp.data.TocData.loadTocs(TocData.java:229)
	at org.eclipse.help.internal.webapp.data.TocData.<init>(TocData.java:119)
	at org.eclipse.help.internal.webapp.data.LayoutData.getContentURL(LayoutData.java:126)
	at org.apache.jsp.advanced.content_jsp._jspService(org.apache.jsp.advanced.content_jsp:135)
	at org.apache.jasper.runtime.HttpJspBase.service(HttpJspBase.java:111)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.apache.jasper.servlet.JspServletWrapper.service(JspServletWrapper.java:403)
	at org.apache.jasper.servlet.JspServlet.serviceJspFile(JspServlet.java:492)
	at org.apache.jasper.servlet.JspServlet.service(JspServlet.java:378)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.jsp.jasper.JspServlet.service(JspServlet.java:155)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.http.registry.internal.ServletManager$ServletWrapper.service(ServletManager.java:180)
	at org.eclipse.equinox.http.servlet.internal.HttpServiceRuntimeImpl$LegacyServlet.service(HttpServiceRuntimeImpl.java:1223)
	at org.eclipse.equinox.http.servlet.internal.registration.EndpointRegistration.service(EndpointRegistration.java:148)
	at org.eclipse.equinox.http.servlet.internal.servlet.ResponseStateHandler.processRequest(ResponseStateHandler.java:62)
	at org.eclipse.equinox.http.servlet.internal.context.DispatchTargets.doDispatch(DispatchTargets.java:131)
	at org.eclipse.equinox.http.servlet.internal.servlet.ProxyServlet.service(ProxyServlet.java:74)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.equinox.http.jetty.internal.HttpServerManager$InternalHttpServiceServlet.service(HttpServerManager.java:284)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:841)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:535)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:188)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:1595)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextHandle(ScopedHandler.java:188)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1253)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:168)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:473)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:1564)
	at org.eclipse.jetty.server.handler.ScopedHandler.nextScope(ScopedHandler.java:166)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1155)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:132)
	at org.eclipse.jetty.server.Server.handle(Server.java:564)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:317)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:251)
	at org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded(AbstractConnection.java:279)
	at org.eclipse.jetty.io.FillInterest.fillable(FillInterest.java:110)
	at org.eclipse.jetty.io.ChannelEndPoint$2.run(ChannelEndPoint.java:124)
	at org.eclipse.jetty.util.thread.Invocable.invokePreferred(Invocable.java:128)
	at org.eclipse.jetty.util.thread.Invocable$InvocableExecutor.invoke(Invocable.java:222)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.doProduce(EatWhatYouKill.java:294)
	at org.eclipse.jetty.util.thread.strategy.EatWhatYouKill.run(EatWhatYouKill.java:199)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:672)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$2.run(QueuedThreadPool.java:590)
	at java.lang.Thread.run(Thread.java:748)

!ENTRY org.eclipse.help 4 0 2017-10-13 11:40:20.109
!MESSAGE Required attribute "label" missing from "topic" element in "/org.yakindu.sct.doc.user/help/user-guide/c-domain-toc.xml" (skipping element)

andreasmuelder, 2017-10-13:
Thx for reporting!

RainerKlute, 2017-10-23:
The _typescript-domain-toc.xml_ issue is fixed with pull request #1724.

I won't fix the _c-domain-toc.xml_ issue, however. That file is _not_ missing. It has just a trifle that is not worth the effort. (At least not at this time. If we'd ever re-write the Eclipse help generator, things will look different for sure.)

closed 2017-10-23 by RainerKlute