itemisCREATE/statecharts #1725: Document Typescript generator

RainerKlute, 2017-10-23:


BeckmaR, 2017-10-23:
The Typescript generator lives in the labs repo now, so maybe move the issue there :)

RainerKlute, 2017-10-23:
The last status I am aware of is that the Typescript generator is in YSCT proper and that the release **must** contain the generator documentation. @andreasmuelder, please comment!

dTrace82, 2017-10-23:
The documentation  is still being processed and can be reviewed tomorrow.

RainerKlute, 2017-10-23:
Word!

dTrace82, 2017-10-24:
@RainerKlute I pushed the first version of the documentation on branch "doc_tsgenerator_issue1725" .  Please review.

RainerKlute, 2017-11-14:
TypeScript generator documentation reviewed.

Please see my changes and notes, the latter being marked with "FIXME". Oh, and please note that the branch's name is "doc_TsGenerator_-issue1725", not "doc_tsgenerator_issue1725".

andreasmuelder, 2018-01-02:
@RainerKlute @dTrace82 can you complete this task

RainerKlute, 2018-01-04:
Hm, I just wanted to review the Typescript generator documentation, but the doc_TsGenerator_-issue1725 branch is still at my last commit 151e5dab3e409180f4ef011db178701431e8069d as of 2017-11-14. @dTrace82, did you forget to push your changes?

andreasmuelder, 2018-03-14:
@dTrace82 can you complete this task

dTrace82, 2018-03-29:
@andreasmuelder @RainerKlute I fixed the most commented issues. A chapter how to integrate the generated code into a web application is currently missing but in preparation.  I have pushed the current state.