itemisCREATE/statecharts #1737: Yakindu  slow down to a crawl on Elementary OS

jcdubois, 2017-10-27:
I am trying to run Yakindu  on elementary OS (https://elementary.io/).

But as soon as I start it the Whole Elementary OS GUI starts slowing down until it is not responding anymore/is frozen.

I tried it on 2 different computers running Elementary OS (a desktop and a laptop: different CPU, different GPU) with the same result.

Note 1: Yakindu is based on eclipse and eclipse is running OK on Elementary OS.

Note 2: Yakindu is running OK on an Ubuntu system.

Note 3: When Yakindu is started the Xserver seems to be getting a lot of CPU load

jcdubois, 2017-10-27:
One solution is to modify the SCT.ini file and adding the 2 following lines before the "-vmargs" line:

`--launcher.GTK_version
2`

This solve the issue.

Should it be on the default SCT.ini file?

andreasmuelder, 2017-10-28:
Good question maybe @BeckmaR can comment on this as I am not a Linux user

BeckmaR, 2017-10-28:
We discussed this already and I believe it‘s in the documentation, but I am not 100% sure. Anyhow, yes, on some systems Eclipse and Gtk3 don‘t work too well together in specific scenarios, depending on the exact gtk3 version for example. There are countless threads about issues. Anyhow, there are systems like Ubuntu where everything runs fine and the latest Eclipse Oxygen has gotten better as well, and gtk2 really doesn‘t look as good as gtk3, so we decided not to use gtk2 as the default option.

I would be interested in the gtk3 version Elementary is running; you can find this out in „Help“->“About“->“Installation Details“, there‘s a long wall of text and somewhere down a few lines this information is noted.

jcdubois, 2017-10-28:
Hi, you can find the content of the Yakindu "configuration" tab in the attached file.

[configuration.txt](https://github.com/Yakindu/statecharts/files/1424272/configuration.txt)

Note that Yakindu was running with "--launcher.GTK_version 2" as it would freeze the entire UI without it.

As for Elementary OS, here is the list of GTK related package I have on my system:

> apport-gtk                       2.20.1-0ubuntu2.10                               
> gir1.2-gtk-3.0:amd64             3.18.9-1ubuntu3.3                                
> gtk2-engines-murrine:amd64       0.98.2-0ubuntu2.2                                
> ibus-gtk:amd64                   1.5.11-1ubuntu2                                  
> ibus-gtk3:amd64                  1.5.11-1ubuntu2                                  
> libcanberra-gtk-module:amd       0.30-2.1ubuntu1                                  
> libcanberra-gtk0:amd64           0.30-2.1ubuntu1                                  
> libcanberra-gtk3-0:amd64         0.30-2.1ubuntu1                                  
> libcanberra-gtk3-module:am       0.30-2.1ubuntu1                                  
> libchamplain-gtk-0.12-0:am       0.12.13-1                                        
> libcheese-gtk25:amd64            3.18.1-2ubuntu3                                  
> libclutter-gtk-1.0-0:amd64       1.6.6-1                                          
> libdbusmenu-gtk3-4:amd64         16.04.1+16.04.20160927-0ubuntu1                  
> libdbusmenu-gtk4:amd64           16.04.1+16.04.20160927-0ubuntu1                  
> libgtk-3-0:amd64                 3.18.9-1ubuntu3.3                                
> libgtk-3-bin                     3.18.9-1ubuntu3.3                                
> libgtk-3-common                  3.18.9-1ubuntu3.3                                
> libgtk2-perl                     2:1.2498-1                                       
> libgtk2.0-0:amd64                2.24.30-1ubuntu1.16.04.2                         
> libgtk2.0-bin                    2.24.30-1ubuntu1.16.04.2                         
> libgtk2.0-common                 2.24.30-1ubuntu1.16.04.2                         
> libgtkglext1:amd64               1.2.0-3.2fakesync1ubuntu1                        
> libgtksourceview-3.0-1:amd       3.18.2-1                                         
> libgtksourceview-3.0-commo       3.18.2-1                                         
> libgtkspell3-3-0:amd64           3.0.7-2                                          
> libjavascriptcoregtk-1.0-0       2.4.11-0ubuntu0.1                                
> libjavascriptcoregtk-3.0-0       2.4.11-0ubuntu0.1                                
> libjavascriptcoregtk-4.0-1       2.18.0-0ubuntu0.16.04.2                          
> libnm-gtk-common                 1.2.6-0ubuntu0.16.04.3+elementary2~ubuntu0.4.1.1 
> libnm-gtk0:amd64                 1.2.6-0ubuntu0.16.04.3+elementary2~ubuntu0.4.1.1 
> libreoffice-gtk                  1:5.1.6~rc2-0ubuntu1~xenial2                     
> libswt-cairo-gtk-3-jni           3.8.2-3                                          
> libswt-glx-gtk-3-jni             3.8.2-3                                          
> libswt-gnome-gtk-3-jni           3.8.2-3                                          
> libswt-gtk-3-java                3.8.2-3                                          
> libswt-gtk-3-jni                 3.8.2-3                                          
> libswt-webkit-gtk-3-jni          3.8.2-3                                          
> libwebkit2gtk-3.0-25:amd64       2.4.11-0ubuntu0.1                                
> libwebkit2gtk-4.0-37:amd64       2.18.0-0ubuntu0.16.04.2                          
> libwebkit2gtk-4.0-37-gtk2:       2.18.0-0ubuntu0.16.04.2                          
> libwebkitgtk-1.0-0:amd64         2.4.11-0ubuntu0.1                                
> libwebkitgtk-1.0-common          2.4.11-0ubuntu0.1                                
> libwebkitgtk-3.0-common          2.4.11-0ubuntu0.1                                
> libwmf0.2-7-gtk                  0.2.8.4-10.5ubuntu1                              
> libwxgtk3.0-0v5:amd64            3.0.2+dfsg-1.3ubuntu0.1                          
> python-gtk2                      2.24.0-4ubuntu1                                  
> python3-aptdaemon.gtk3widg       1.1.1+bzr982-0ubuntu14                           
> xdg-user-dirs-gtk                0.10-1ubuntu1+elementary2~ubuntu16.04.1

Note that there might be more GTK packages/versions than the default install as I have added some in my search for a solution.

Tell me if you want more information.

JC

BeckmaR, 2017-11-03:
Hey,

thank you for the input. Weirdly, it looks like the libgtk3 version of yours would be the exact same as mine on my Ubuntu system. 

The next thing you could check is your video driver. Please run `lshw -c video` (no need for sudo) and check the "Configuration" part for "driver". Then you could proceed to find out the version of this driver you have installed, and check if there are any known problems with it or with this gtk3 version or your gpu in general.

jcdubois, 2017-11-05:
Here is the output for one of the 2 systems:

$ lshw -c video
ATTENTION: ce programme devrait être lancé en tant que super-utilisateur.
  *-display:0             
       description: VGA compatible controller
       produit: 4 Series Chipset Integrated Graphics Controller
       fabriquant: Intel Corporation
       identifiant matériel: 2
       information bus: pci@0000:00:02.0
       version: 03
       bits: 64 bits
       horloge: 33MHz
       fonctionnalités: vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       ressources: irq:28 mémoire:fe400000-fe7fffff mémoire:e0000000-efffffff portE/S:cc00(taille=8) mémoire:c0000-dffff
  *-display:1 NON-RÉCLAMÉ
       description: Display controller
       produit: 4 Series Chipset Integrated Graphics Controller
       fabriquant: Intel Corporation
       identifiant matériel: 2.1
       information bus: pci@0000:00:02.1
       version: 03
       bits: 64 bits
       horloge: 33MHz
       fonctionnalités: bus_master cap_list
       configuration: latency=0
       ressources: mémoire:fe900000-fe9fffff

So driver is i915 and kernel is 4.10.0-37 generic (I assume it is modern enough).

The second system is owned by my daughter and I am waiting for her feed back. But her system is based on core i5 so the driver should also be i915 I believe.

jcdubois, 2017-11-08:
Just to confirm that the second system is also running i915 driver.

*-display               
       description: VGA compatible controller
       produit: Intel Corporation
       fabriquant: Intel Corporation
       identifiant matériel: 2
       information bus: pci@0000:00:02.0
       version: 02
       bits: 64 bits
       horloge: 33MHz
       fonctionnalités: pciexpress msi pm vga_controller bus_master cap_list rom
       configuration: driver=i915 latency=0
       ressources: irq:126 mémoire:ee000000-eeffffff mémoire:d0000000-dfffffff portE/S:f000(taille=64) mémoire:c0000-dffff

BeckmaR, 2017-11-10:
So, what I can read from your first system is that it apparently isn't exactly new. However, SCT isn't really graphically complex and should work on older machines, too. As I understand it, the first system uses Elementary and the second one Ubuntu? Or did you try both on both machines?

For now, I can only suspect an issue with Elementary's desktop environment or the compositor, maybe something is buggy in there in relation to GTK3.

However, you are completely right that our documentation is missing a part about forcing Eclipse to use GTK2. I will add it.

jcdubois, 2017-11-10:
Both systems are running elementary-OS (the old and the new) both systems are using i915 driver and the same linux kernel version and both systems are freezing (after a while) when Yakindu is started without the GTK2 workaround.

I got another system (a core m5) under unbuntu 17.10 that has no issue with Yakindu without the workaround.

commit a726deb2 (2017-11-16): Add section about using GTK2 instead of GTK3 - close #1737 (#1741)
closed 2017-11-16 by andreasmuelder