itemisCREATE/statecharts #1739: Add increment / decrement expression

andreasmuelder, 2017-11-05:
x++, ++x, --x, x-- are not possible yet

BeckmaR, 2017-11-10:
We should add validation for expressions that use the variable and a post-in/decrement, as their behavior is not defined and might return something else when compiled then what we simulate.

See this reddit thread for more information and a live example:
[Please explain why the result of ++a + ++a + ++a is equal to 13? I don't quite understand how it's working.](https://www.reddit.com/r/programming/comments/7byved/please_explain_why_the_result_of_a_a_a_is_equal/)

andreasmuelder, 2017-11-30:
We should implement postfix increment / decrement at a first step.

tkutz, 2017-12-05:
Postfix Expressions are now available in master. 
Shall we close that ticket, or keep it until prefix expressions are implemented?
Or do we postpone implementation of prefix increment/decrement due to aforementioned ambiguities with C?

BeckmaR, 2017-12-05:
Actually, the ambiguities exist both with postfix and prefix. 

René Beckmann

andreasmuelder, 2017-12-11:
if prefix increment decrement is requried we should open a new ticket

closed 2017-12-11 by andreasmuelder