itemisCREATE/statecharts #1770: Place Version Info into generated Code

DefCon-CC, 2017-11-29:
It would be great when Yakindu SCT could place Version Info (Yakindu SCT Version, ....) into generated code.

BeckmaR, 2017-12-11:
This is a good idea! The linked PR contains new builtin variables which you can refer to in your sgen file. This allows you to put the current SCT version, the timestamp, the user's name and the hostname into whatever string you want to.

For example, this sgen:
```
GeneratorModel for yakindu::cpp {

	statechart statechart {

		feature Outlet {
			targetProject = "Project"
			targetFolder = "src-gen"
			libraryTargetFolder = "src"
		}
		feature LicenseHeader {
			licenseText = "Generated on " + TIMESTAMP + " with YAKINDU SCT " + SCTVERSION + " by " + USER + " on " + HOSTNAME
		}
	}
}
```

generates the following license header into the generated files:
```
/* Generated on 11.12.2017 17:08:14 with YAKINDU SCT 3.2.0 by BeckmaR on BeckmaR */
```

DefCon-CC, 2017-12-12:
Great. Thanks!

andreasmuelder, 2018-01-10:
As commented by @RainerKlute we should use ISO-8601-Format for the date: YYYY-MM-TT.

BeckmaR, 2018-01-11:
Closing as discussed with @andreasmuelder - the current date format is simply localized, as stated in the documentation.

PR #1786 (2018-01-03): Issue 1770 - sgen variables
Introduces builtin variables to the sgen expressions.

Users can use the following strings in sgens:

* `TIMESTAMP` prints the current date and time in the machine's locale.
* `SCTVERSION` prints the current SCT version in format `3.2.0` 
* `USER` prints the currently logged in user's name (`System.getProperty("user.name")`)
* `HOSTNAME` prints the name of the machine on which the code was generated.

This can for example be used together with the licenseheader feature to embed generation info into the generated files.

Resolves #1770 
Resolves #1787
closed 2018-01-11 by BeckmaR