itemisCREATE/statecharts #1783: Installation problem from update site with MacOS and Eclipse Neon

tkutz, 2017-12-11:
Steps to reproduce:
Install Eclipse Neon.3, e.g. from here: 
http://www.eclipse.org/downloads/download.php?file=/technology/epp/downloads/release/neon/3/eclipse-java-neon-3-macosx-cocoa-x86_64.tar.gz

Install SCT Standard Edition from Update Site.

Following error appears:

[image]

tkutz, 2017-12-11:
It works when I put the Eclipse instance into my Applications folder.

I know this was not the case in previous versions but might be due to an update of my OS.

svenjawendler, 2018-01-11:
This problem also occures on Mac OS Sierra 10.12.4. When eclipse.app lies in app folder, everything works fine.

svenjawendler, 2018-01-11:
Shall we add a remark on the installation documentation section? Like: please ensure that your eclipse application is situated within the app folder?

tkutz, 2018-01-11:
See also following articles describing the same problem:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=491326
https://stackoverflow.com/questions/25091732/eclipse-installing-software-chmod-failed

closed 2018-01-16 by svenjawendler