itemisCREATE/statecharts #1794: 3.2.0 Test - DefinitionSection: if it is pinned no errors, warnings are shown

svenjawendler, 2017-12-13:
On Windows: If the definitions section is pinned neither the diagram nor the problems view shows any warnings or errors, though there are problems.
If I unpin it again the errors/warnings come back.
[image]

svenjawendler, 2017-12-13:
A clean project seems to solve the problem. If I clean the project the markers are there. If I pin the definition section and solve the warnings errors the problems view is not updated. A clean project solves this.

svenjawendler, 2017-12-13:
Another problem: When the definition section is in pinned state the pin icon seems to overlap the error marker. Fact is I do not see any error marker than. In the former version the markers were at the same position like the pin icon now is.
[image]

andreasmuelder, 2018-01-02:
@svenjawendler should be fixed, please reopen if the error still exists

closed 2018-01-02 by andreasmuelder