itemisCREATE/statecharts #1812: [SCTUnit] - Cannot run elevator example SCTUnit Test, IllegalArgumentException

svenjawendler, 2017-12-14:
java.lang.IllegalArgumentException
	at org.eclipse.ui.part.FileEditorInput.getPath(FileEditorInput.java:181)
	at org.eclipse.ui.internal.WorkbenchPage.busyOpenEditor(WorkbenchPage.java:3190)
	at org.eclipse.ui.internal.WorkbenchPage.access$25(WorkbenchPage.java:3176)
	at org.eclipse.ui.internal.WorkbenchPage$10.run(WorkbenchPage.java:3158)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:70)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3153)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3117)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3107)
	at org.eclipse.ui.ide.IDE.openEditor(IDE.java:563)
	at org.eclipse.ui.ide.IDE.openEditor(IDE.java:518)
	at com.yakindu.sctunit.simulation.ui.handler.OpenResultStatusHandler$1.run(OpenResultStatusHandler.java:38)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:37)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:182)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4213)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3820)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1150)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1039)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:680)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:594)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:151)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)

rherrmannr, 2017-12-14:
can agree,

it's not working in 3.1.1 and 3.2.0 release candidate.

rherrmannr, 2017-12-14:
The locationUri cannot be found.

It only occurs if the example wizard is used.
It's possible to copy the model into a new project and create a new sctunit testclass. The test will be executed.

BeckmaR, 2017-12-21:
This is a professional edition issue.

closed 2017-12-21 by BeckmaR