itemisCREATE/statecharts #1841: Changing font color in Yakindu SCT preferences has no effect

WolfgangHa, 2018-01-05:
Detected in YAKINDU Statechart Tools Standard Edition version 3.1.1

Open YAKINDU SCT and open the "Preferences" dialog.
Select the node "YAKINDU SCT -> Diagram Appearance".
Press the button "Change..." in the "Font styles" section.
In the FontDialog select any other Color, e.g. "Red" and press "OK".
Press the button "Change..." in the "Font styles" section again.
OBSERVED:
-> The former color (black) is selected
EXPECTED:
-> The changed color (red) is selected.

andreasmuelder, 2018-01-05:
@WolfgangHa I can reproduce this. 
The font can be changed, changing the color does not have any effect.

rherrmannr, 2018-01-10:
Additionally, there is no option to change the font color in Ubuntu.

andreasmuelder, 2018-02-05:
I had a look into this. The dialog to change the font style is platform specific and provided by the underlying OS.  So we can not use the specified color since this option is not available for Linux / MacOS.

Unfortunately we can not do much here.

closed 2018-02-05 by andreasmuelder