itemisCREATE/statecharts #1877: 3.1.1 Stack Trace in reaction to C++

emrainey, 2018-01-24:
I've been using a 3.1.1 version to do some unsupported things (Deep C++ integration) and when it failed it had an unexpectedly large and repetitive failure which I didn't expect. This sort of stacktrace was repeated thousands of times and made it very hard to understand what was really going wrong until I realized that my Yakindu version didn't have the C++ integration that the website docs talked about. The bug, I think, is that this trace was unhelpful in understanding the problem. 

> Jan 24, 2018 9:41:27 AM org.yakindu.sct.model.stext.scoping.STextScopeProvider$ErrorHandlerDelegate handle
WARNING: Error in scope method, using fallback
java.lang.NullPointerException
	at com.yakindu.sct.domain.c.runtime.resource.CHeaderPackageImport2URIMapper.getAllImports(CHeaderPackageImport2URIMapper.java:48)
	at com.yakindu.sct.domain.c.runtime.resource.CHeaderPackageImport2URIMapper.findPackageImport(CHeaderPackageImport2URIMapper.java:37)
	at com.yakindu.sct.domain.c.runtime.resource.CHeaderUriMapperComposite.findPackageImport(CHeaderUriMapperComposite.java:31)
	at org.yakindu.sct.model.stext.scoping.STextGlobalScopeProvider.collectPackageImports(STextGlobalScopeProvider.java:161)
	at org.yakindu.sct.model.stext.scoping.STextGlobalScopeProvider$4.get(STextGlobalScopeProvider.java:135)
	at org.yakindu.sct.model.stext.scoping.STextGlobalScopeProvider$4.get(STextGlobalScopeProvider.java:1)
	at org.eclipse.xtext.util.OnChangeEvictingCache.get(OnChangeEvictingCache.java:77)
	at org.yakindu.sct.model.stext.scoping.STextGlobalScopeProvider.getImportedUris(STextGlobalScopeProvider.java:126)
	at org.yakindu.sct.model.stext.scoping.STextGlobalScopeProvider.getScope(STextGlobalScopeProvider.java:185)
	at org.eclipse.xtext.scoping.impl.AbstractGlobalScopeProvider.getScope(AbstractGlobalScopeProvider.java:50)
	at org.yakindu.sct.model.stext.scoping.STextGlobalScopeProvider.getScope(STextGlobalScopeProvider.java:86)
	at org.eclipse.xtext.scoping.impl.AbstractGlobalScopeDelegatingScopeProvider.getGlobalScope(AbstractGlobalScopeDelegatingScopeProvider.java:46)
	at org.eclipse.xtext.scoping.impl.AbstractGlobalScopeDelegatingScopeProvider.getGlobalScope(AbstractGlobalScopeDelegatingScopeProvider.java:42)
	at org.eclipse.xtext.scoping.impl.ImportedNamespaceAwareLocalScopeProvider.getResourceScope(ImportedNamespaceAwareLocalScopeProvider.java:104)
	at org.yakindu.sct.model.stext.scoping.StextImportAwareScopeProvider.getScope(StextImportAwareScopeProvider.java:58)
	at org.yakindu.sct.model.stext.scoping.STextScopeProvider.scope_TypeSpecifier_type(STextScopeProvider.java:143)
	at sun.reflect.GeneratedMethodAccessor5.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.xtext.util.PolymorphicDispatcher.invoke(PolymorphicDispatcher.java:302)
	at org.eclipse.xtext.scoping.impl.AbstractDeclarativeScopeProvider.polymorphicFindScopeForReferenceName(AbstractDeclarativeScopeProvider.java:152)
	at org.eclipse.xtext.scoping.impl.AbstractDeclarativeScopeProvider.getScope(AbstractDeclarativeScopeProvider.java:100)
	at org.yakindu.sct.model.stext.scoping.STextScopeProvider.getScope(STextScopeProvider.java:103)
	at org.eclipse.xtext.linking.impl.DefaultLinkingService.getScope(DefaultLinkingService.java:59)
	at org.eclipse.xtext.linking.impl.DefaultLinkingService.getLinkedObjects(DefaultLinkingService.java:120)
	at org.yakindu.sct.model.sgraph.resource.AbstractSCTResource.getEObject(AbstractSCTResource.java:184)
	at org.eclipse.emf.ecore.resource.impl.ResourceSetImpl.getEObject(ResourceSetImpl.java:223)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:205)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolve(EcoreUtil.java:261)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eResolveProxy(BasicEObjectImpl.java:1477)
	at org.yakindu.base.types.impl.DeclarationImpl.getType(DeclarationImpl.java:121)
	at org.yakindu.base.types.impl.DeclarationImpl.eGet(DeclarationImpl.java:237)
	at org.yakindu.base.types.impl.PropertyImpl.eGet(PropertyImpl.java:183)
	at org.yakindu.sct.model.stext.stext.impl.VariableDefinitionImpl.eGet(VariableDefinitionImpl.java:126)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eGet(BasicEObjectImpl.java:1011)
	at org.eclipse.emf.ecore.impl.BasicEObjectImpl.eGet(BasicEObjectImpl.java:1003)
	at org.eclipse.emf.ecore.util.EContentsEList$FeatureIteratorImpl.hasNext(EContentsEList.java:439)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolveCrossReferences(EcoreUtil.java:306)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolveAll(EcoreUtil.java:300)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolveAll(EcoreUtil.java:285)
	at org.eclipse.emf.ecore.util.EcoreUtil.resolveAll(EcoreUtil.java:273)
	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.createExecutor(GeneratorExecutorLookup.java:70)
	at org.yakindu.sct.generator.core.execution.GeneratorExecutorLookup.execute(GeneratorExecutorLookup.java:49)
	at com.yakindu.sct.generator.application.GeneratorApplication.execGenerator(GeneratorApplication.java:505)
	at com.yakindu.sct.generator.application.GeneratorApplication.generateAll(GeneratorApplication.java:417)
	at com.yakindu.sct.generator.application.GeneratorApplication.start(GeneratorApplication.java:136)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)

robert-rudi, 2018-01-25:
Hello @emrainey, thanks for this report. Is this exception reproducible for your usecases in Yakindu Statechart Tools v3.2.0?

BeckmaR, 2018-01-26:
Hey @emrainey, the website doc always reflects the newest version YAKINDU Statechart Tools. When in doubt, use the help that is available from inside the tool, via [`Help`] -> [`Help Contents`].

The trace you are seeing was not helpful because it is an unexpected error, and it's probably quite long because it is in the ScopeProvider where the same piece of code is probably called a thousand times. I agree with you that software should always provide error messages as meaningful as possible - however that is only doable in expected situations, i.e. when you try to simulate a statechart which contains errors, or when the generator model references a statechart it can't find, stuff like that.

In this case, you tried to access a feature which was not ready for production and was only available as a side effect. The feature is now available as a beta version in the professional edition.

I will close the issue for now, because the issue should be fixed in the current version and does not contain any steps to reproduce. If you encounter any issue in the new version or you re-encounter this issue, please feel free to re-open this issue or to create a new one at any time. We are always happy about feedback from our users!

closed 2018-01-26 by BeckmaR