itemisCREATE/statecharts #1881: Crash in 3.1.1 on Startup

emrainey, 2018-01-26:
First I installed some common Eclipse plugins into the SCTPRO. (DLTK for Shell/Ruby, CMake, Color Theme). After this, my command line "headless" invocation of scc failed. I elided the paths and file names with ...

> "/Applications/YAKINDU SCTPRO.app/Contents/Eclipse/scc" -m /.../ -v "headless=true" -v "output_folder=/.../" -v "project_name=..."
/Applications/YAKINDU SCTPRO.app/Contents/Eclipse/scc: line 42: 84946 Segmentation fault: 11  "$SCTEXECUTABLE" --launcher.ini "$SCRIPTPATH/headless.ini" "$@" $BASEDIR_ARG -vmargs -Dorg.eclipse.emf.ecore.plugin.EcorePlugin.doNotLoadResourcesPlugin=true
.../....d:10: *** [....h] Error 139

I tried trapping in lldb to see what's the problem...

> lldb /Applications/YAKINDU\ SCTPRO.app/Contents/MacOS/SCT
(lldb) target create "/Applications/YAKINDU SCTPRO.app/Contents/MacOS/SCT"
Current executable set to '/Applications/YAKINDU SCTPRO.app/Contents/MacOS/SCT' (x86_64).
(lldb) run --launcher.ini /Applications/YAKINDU\ SCTPRO.app/Contents/Eclipse/headless.ini -vmargs -Dorg.eclipse.emf.ecore.plugin.EcorePlugin.doNotLoadResourecePlugin=true
Process 90666 launched: '/Applications/YAKINDU SCTPRO.app/Contents/MacOS/SCT' (x86_64)
2018-01-26 10:18:51.543993-0800 SCT[90666:1203591] WARNING: The Gestalt selector gestaltSystemVersion is returning 10.9.6 instead of 10.12.6. This is not a bug in Gestalt -- it is a documented limitation. Use NSProcessInfo's operatingSystemVersion property to get correct system version number.
Call location:
2018-01-26 10:18:51.544208-0800 SCT[90666:1203591] 0   CarbonCore                          0x00007fffaf1d01b1 ___Gestalt_SystemVersion_block_invoke + 121
2018-01-26 10:18:51.544230-0800 SCT[90666:1203591] 1   libdispatch.dylib                   0x00007fffc3a178fc _dispatch_client_callout + 8
2018-01-26 10:18:51.544237-0800 SCT[90666:1203591] 2   libdispatch.dylib                   0x00007fffc3a178b9 dispatch_once_f + 38
2018-01-26 10:18:51.544244-0800 SCT[90666:1203591] 3   CarbonCore                          0x00007fffaf15e292 _Gestalt_SystemVersion + 976
2018-01-26 10:18:51.544251-0800 SCT[90666:1203591] 4   CarbonCore                          0x00007fffaf15daad Gestalt + 139
2018-01-26 10:18:51.544258-0800 SCT[90666:1203591] 5   SCT                                 0x0000000100001cc3 main + 67
2018-01-26 10:18:51.544264-0800 SCT[90666:1203591] 6   libdyld.dylib                       0x00007fffc3a4d235 start + 1
Process 90666 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x3c)
    frame #0: 0x00007fffae32bb74 CoreFoundation`CFBundleGetFunctionPointerForName + 36
CoreFoundation`CFBundleGetFunctionPointerForName:
->  0x7fffae32bb74 <+36>: cmpb   $0x0, 0x3c(%rbx)
    0x7fffae32bb78 <+40>: jne    0x7fffae32bb8a            ; <+58>
    0x7fffae32bb7a <+42>: movq   %rbx, %rdi
    0x7fffae32bb7d <+45>: callq  0x7fffae305b20            ; CFBundleLoadExecutable
Target 0: (SCT) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x3c)
  * frame #0: 0x00007fffae32bb74 CoreFoundation`CFBundleGetFunctionPointerForName + 36
    frame #1: 0x00000001000035a2 SCT`findSymbol + 66
    frame #2: 0x0000000100001910 SCT`original_main + 1296
    frame #3: 0x00000001000022e5 SCT`main + 1637
    frame #4: 0x00007fffc3a4d235 libdyld.dylib`start + 1

I even made a second install in a different location and that still failed in the same way. Is this a known issue in 3.1.1?

emrainey, 2018-01-26:
I've also deleted every eclipse related "dotfolder" in the folder I'm trying to call this in. Still crashes.

emrainey, 2018-01-27:
I installed to 3.2.0 and had a different failure mode. An alert came up that said:

I hide the project names and paths with "..."

> Java was started but returned exit code=-1
> -Dorg.eclipse.emf.ecore.plugin.EcorePlugin.doNotLoadResourcesPlugin=true
> -Djava.class.path=/Applications/YAKINDU SCTPRO.app/Contents/MacOS//../Eclipse/plugins/org.eclipse.equinox.launcher_1.4.0.v20161219-1356.jar
> -os macosx
> -ws cocoa
> -arch x86_64
> -launcher /Applications/YAKINDU SCTPRO.app/Contents/MacOS/SCT
> -name SCT
> --launcher.library /Applications/YAKINDU SCTPRO.app/Contents/MacOS//../Eclipse/plugins/org.eclipse.equinox.launcher.cocoa.macosx.x86_64_1.1.551.v20171108-1834/eclipse_1630.so
> -startup /Applications/YAKINDU SCTPRO.app/Contents/MacOS//../Eclipse/plugins/org.eclipse.equinox.launcher_1.4.0.v20161219-1356.jar
> --launcher.overrideVmargs
> -application com.yakindu.sct.generator.application
> -data @noDefault
> -eclipse.product org.eclipse.platform.ide
> -m /.../
> -v headless=true
> -v output_folder=.../
> -v project_name=...
> -d "/..."
> -vm /Library/Java/JavaVirtualMachines/jdk1.8.0_151.jdk/Contents/Home/bin/../jre/lib/server/libjvm.dylib
> -vmargs
> -Dorg.eclipse.emf.ecore.plugin.EcorePlugin.doNotLoadResourcesPlugin=true
> -Djava.class.path=/Applications/YAKINDU SCTPRO.app/Contents/MacOS//../Eclipse/plugins/org.eclipse.equinox.launcher_1.4.0.v20161219-1356.jar

In the command line I got this: 

> "/Applications/YAKINDU SCTPRO.app/Contents/Eclipse/scc" -m /.../ -v "headless=true" -v "output_folder=.../" -v "project_name=..."
> --------------------------------------------------------------
> YAKINDU Statechart Tools Headless Generator ((c) by itemis AG)
> 
> 	  Visit http://www.statecharts.org
> --------------------------------------------------------------
> Loading models from: /Applications/YAKINDU SCTPRO.app/Contents/MacOS
> 0    [main] ERROR com.yakindu.sct.generator.application.GeneratorApplication  - Error
> com.yakindu.sct.generator.application.GeneratorApplication$IllegalSctHeadlessArgumentException: Cannot find any .project files in : [/.../]
> 	at com.yakindu.sct.generator.application.ProjectImporter.importProjects(ProjectImporter.java:47)
> 	at com.yakindu.sct.generator.application.GeneratorApplication.loadModels(GeneratorApplication.java:252)
> 	at com.yakindu.sct.generator.application.GeneratorApplication.start(GeneratorApplication.java:150)
> 	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
> 	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
> 	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
> 	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
> 	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.lang.reflect.Method.invoke(Method.java:498)
> 	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
> 	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
> 	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
> 0 [main] ERROR com.yakindu.sct.generator.application.GeneratorApplication  - Error
> com.yakindu.sct.generator.application.GeneratorApplication$IllegalSctHeadlessArgumentException: Cannot find any .project files in : [/.../]
> 	at com.yakindu.sct.generator.application.ProjectImporter.importProjects(ProjectImporter.java:47)
> 	at com.yakindu.sct.generator.application.GeneratorApplication.loadModels(GeneratorApplication.java:252)
> 	at com.yakindu.sct.generator.application.GeneratorApplication.start(GeneratorApplication.java:150)
> 	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
> 	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
> 	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
> 	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
> 	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
> 	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
> 	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
> 	at java.lang.reflect.Method.invoke(Method.java:498)
> 	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
> 	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
> 	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
> 

There definitely are not any .project files at the locations it was complaining about and there aren't supposed to be either. Just the .sct and .sgen files. Are .project files *required* now?

tkutz, 2018-01-29:
Let me try to shed some light into this one.

Your first problem (segmentation fault) might indicate that the macOS launcher plugin version as specified in headless.ini differs from the installed one. If that worked before, this might be a side-effect from updating your SCT Eclipse. This is a problem specifically to macOS, as only here we need to explicitly specify the macOS launcher plugin to be used. Of course, this is not optimal. A workaround is to open headless.ini and adjust the line
```
--launcher.library
../Eclipse/plugins/org.eclipse.equinox.launcher.cocoa.macosx.x86_64_1.1.550.v20170928-1359
```
with the concrete version that is installed under `SCT.app/Contents/Eclipse/plugins`

Regarding the second problem, yes, it is required to have .project files when using scc. The internal logic of the headless execution derives relevant information from .project files.

emrainey, 2018-01-30:
The existing 3.1.1 headless.ini had this:

```
--launcher.library
../Eclipse/plugins/org.eclipse.equinox.launcher.cocoa.macosx.x86_64_1.1.500.v20170531-1133
```

When I changed it to what you mentioned:

```
--launcher.library
../Eclipse/plugins/org.eclipse.equinox.launcher.cocoa.macosx.x86_64_1.1.550.v20170928-1359
```

It stopped crashing and worked correctly. Looks like the file needs to be updated in the 3.1.1 Mac OSX DMG.

BeckmaR, 2018-02-01:
Hey @emrainey, I'm sorry to hear you ran into so many problems. As far as I can see, your main issue - the wrong launcher version in the headless.ini - was solved. We already have an issue for that in our internal issue tracker, so I'd close this issue now. If you still have questions or if you run into more problems, please feel free to reopen this issue or open a new issue all the time.

If you do have a license for the Professional Edition, you can also use the support button integrated therein, to directly send an email to our support team.

closed 2018-02-01 by BeckmaR