itemisCREATE/statecharts #1905: Deadlock on Safe / Validation

BeckmaR, 2018-02-09:
Occasionally there's a deadlock on saving the statechart. The two conflicting threads are the main thread which wants to save the workspace:

```
Thread [main] (Suspended)	
	owns: WorkspaceModifyDelegatingOperation  (id=101)	
	waiting for: Object  (id=100)	
	Object.wait(long) line: not available [native method]	
	ThreadJob.waitForRun(ThreadJob, IProgressMonitor, InternalJob, Thread) line: 272	
	ThreadJob.joinRun(ThreadJob, IProgressMonitor) line: 197	
	ImplicitJobs.begin(ISchedulingRule, IProgressMonitor, boolean) line: 92	
	JobManager.beginRule(ISchedulingRule, IProgressMonitor) line: 307	
	WorkManager.checkIn(ISchedulingRule, IProgressMonitor) line: 121	
	Workspace.prepareOperation(ISchedulingRule, IProgressMonitor) line: 2188	
	Workspace.run(ICoreRunnable, ISchedulingRule, int, IProgressMonitor) line: 2235	
	Workspace.run(IWorkspaceRunnable, ISchedulingRule, int, IProgressMonitor) line: 2267	
	WorkspaceModifyDelegatingOperation(WorkspaceModifyOperation).run(IProgressMonitor) line: 128	
	WorkspaceOperationRunner.run(boolean, boolean, IRunnableWithProgress, ISchedulingRule) line: 75	
	WorkspaceOperationRunner.run(boolean, boolean, IRunnableWithProgress) line: 65	
	DiagramPartitioningDocumentProvider(AbstractDocumentProvider).executeOperation(AbstractDocumentProvider$DocumentProviderOperation, IProgressMonitor) line: 528	
	DiagramPartitioningDocumentProvider(AbstractDocumentProvider).saveDocument(IProgressMonitor, Object, IDocument, boolean) line: 579	
	StatechartDiagramEditor(DiagramDocumentEditor).performSave(boolean, IProgressMonitor) line: 864	
	StatechartDiagramEditor(DiagramDocumentEditor).doSave(IProgressMonitor) line: 684	
	SaveableHelper$2.run(IProgressMonitor) line: 157	
	SaveableHelper$5.run(IProgressMonitor) line: 286	
	ModalContext.runInCurrentThread(IRunnableWithProgress, IProgressMonitor) line: 437	
	ModalContext.run(IRunnableWithProgress, boolean, IProgressMonitor, Display) line: 353	
	WorkbenchWindow$14.run() line: 2184	
	BusyIndicator.showWhile(Display, Runnable) line: 70	
	WorkbenchWindow.run(boolean, boolean, IRunnableWithProgress) line: 2180	
	SaveableHelper.runProgressMonitorOperation(String, IRunnableWithProgress, IRunnableContext, IShellProvider) line: 294	
	SaveableHelper.runProgressMonitorOperation(String, IRunnableWithProgress, IWorkbenchWindow) line: 272	
	SaveableHelper.savePart(ISaveablePart, IWorkbenchPart, IWorkbenchWindow, boolean) line: 162	
	WorkbenchPage.saveSaveable(ISaveablePart, IWorkbenchPart, boolean, boolean) line: 3904	
	WorkbenchPage.saveEditor(IEditorPart, boolean) line: 3918	
	SaveHandler.execute(ExecutionEvent) line: 54	
	HandlerProxy.execute(ExecutionEvent) line: 295	
	E4HandlerProxy.execute(IEclipseContext, Map, Event, IEvaluationContext) line: 90	
	NativeMethodAccessorImpl.invoke0(Method, Object, Object[]) line: not available [native method]	
	NativeMethodAccessorImpl.invoke(Object, Object[]) line: 62	
	DelegatingMethodAccessorImpl.invoke(Object, Object[]) line: 43	
	Method.invoke(Object, Object...) line: 498	
	MethodRequestor.execute() line: 55	
	InjectorImpl.invokeUsingClass(Object, Class<?>, Class<Annotation>, Object, PrimaryObjectSupplier, PrimaryObjectSupplier, boolean) line: 282	
	InjectorImpl.invoke(Object, Class<Annotation>, Object, PrimaryObjectSupplier, PrimaryObjectSupplier) line: 264	
	ContextInjectionFactory.invoke(Object, Class<Annotation>, IEclipseContext, IEclipseContext, Object) line: 132	
	WorkbenchHandlerServiceHandler(HandlerServiceHandler).execute(ExecutionEvent) line: 152	
	Command.executeWithChecks(ExecutionEvent) line: 494	
	ParameterizedCommand.executeWithChecks(Object, Object) line: 488	
	HandlerServiceImpl.executeHandler(ParameterizedCommand, IEclipseContext) line: 210	
	KeyBindingDispatcher.executeCommand(ParameterizedCommand, Event) line: 286	
	KeyBindingDispatcher.press(List<KeyStroke>, Event) line: 507	
	KeyBindingDispatcher.processKeyEvent(List<KeyStroke>, Event) line: 558	
	KeyBindingDispatcher.filterKeySequenceBindings(Event) line: 378	
	KeyBindingDispatcher.access$0(KeyBindingDispatcher, Event) line: 324	
	KeyBindingDispatcher$KeyDownFilter.handleEvent(Event) line: 86	
	EventTable.sendEvent(Event) line: 84	
	Display.filterEvent(Event) line: 1605	
	FigureCanvas(Widget).sendEvent(Event) line: 1339	
	FigureCanvas(Widget).sendEvent(int, Event, boolean) line: 1366	
	FigureCanvas(Widget).sendEvent(int, Event) line: 1349	
	FigureCanvas(Widget).sendKeyEvent(int, GdkEventKey) line: 1378	
	FigureCanvas(Widget).gtk_key_press_event(long, long) line: 764	
	FigureCanvas(Control).gtk_key_press_event(long, long) line: 3465	
	FigureCanvas(Composite).gtk_key_press_event(long, long) line: 801	
	FigureCanvas(Widget).windowProc(long, long, long) line: 2000	
	FigureCanvas(Control).windowProc(long, long, long) line: 5827	
	Display.windowProc(long, long, long) line: 5498	
	OS._gtk_main_do_event(long) line: not available [native method]	
	OS.gtk_main_do_event(long) line: 9545	
	Display.eventProc(long, long) line: 1275	
	OS._g_main_context_iteration(long, boolean) line: not available [native method]	
	OS.g_main_context_iteration(long, boolean) line: 2495	
	Display.readAndDispatch() line: 4149	
	PartRenderingEngine$4.run() line: 1121	
	Realm.runWithDefault(Realm, Runnable) line: 336	
	PartRenderingEngine.run(MApplicationElement, IEclipseContext) line: 1022	
	E4Workbench.createAndRunUI(MApplicationElement) line: 150	
	Workbench$5.run() line: 693	
	Realm.runWithDefault(Realm, Runnable) line: 336	
	Workbench.createAndRunWorkbench(Display, WorkbenchAdvisor) line: 610	
	PlatformUI.createAndRunWorkbench(Display, WorkbenchAdvisor) line: 148	
	IDEApplication.start(IApplicationContext) line: 138	
	EclipseAppHandle.run(Object) line: 196	
	EclipseAppLauncher.runApplication(Object) line: 134	
	EclipseAppLauncher.start(Object) line: 104	
	EclipseStarter.run(Object) line: 388	
	EclipseStarter.run(String[], Runnable) line: 243	
	NativeMethodAccessorImpl.invoke0(Method, Object, Object[]) line: not available [native method]	
	NativeMethodAccessorImpl.invoke(Object, Object[]) line: 62	
	DelegatingMethodAccessorImpl.invoke(Object, Object[]) line: 43	
	Method.invoke(Object, Object...) line: 498	
	Main.invokeFramework(String[], URL[]) line: 673	
	Main.basicRun(String[]) line: 610	
	Main.run(String[]) line: 1519	
	Main.main(String[]) line: 1492	
```

and the changed issueStore, which wants to paint the issue decorators:

```
Thread [Worker-6] (Suspended)	
	owns: ArrayList<E>  (id=94)	
	owns: DefaultValidationIssueStore  (id=95)	
	waiting for: Lock  (id=93)	
	Object.wait(long) line: not available [native method]	
	Lock(Object).wait() line: 502 [local variables unavailable]	
	Lock.lock() line: 34	
	OS.g_utf16_to_utf8(char[], long, long[], long[], long[]) line: 3579	
	Converter.wcsToMbcs(String, char[], boolean) line: 63	
	Image.initNative(String) line: 913	
	Image.<init>(Device, ImageFileNameProvider) line: 777	
	URLImageDescriptor.createImage(boolean, Device) line: 244	
	URLImageDescriptor(ImageDescriptor).createResource(Device) line: 164	
	DeviceResourceManager.allocate(DeviceResourceDescriptor) line: 52	
	DeviceResourceManager(AbstractResourceManager).create(DeviceResourceDescriptor) line: 86	
	DeviceResourceManager(ResourceManager).createImageWithDefault(ImageDescriptor) line: 192	
	ImageRegistry.get(String) line: 204	
	WorkbenchImages.getImage(String) line: 430	
	SharedImages.getImage(String) line: 28	
	StatechartValidationDecorationProvider$ValidationDecorator.getImage(Severity) line: 233	
	StatechartValidationDecorationProvider$ValidationDecorator.decorate(View) line: 163	
	StatechartValidationDecorationProvider$ValidationDecorator.refresh() line: 121	
	StatechartValidationDecorationProvider$ValidationDecorator.issuesChanged() line: 238	
	DefaultValidationIssueStore.notifyListeners(String) line: 98	
	DefaultValidationIssueStore.processIssues(List<Issue>, IProgressMonitor) line: 199	
	ShadowModelValidationJob.runInternal(IProgressMonitor) line: 68	
	ShadowModelValidationJob(ValidationJob).run(IProgressMonitor) line: 46	
	Worker.run() line: 55	
```

andreasmuelder, 2018-03-12:
@BeckmaR how to reproduce?

BeckmaR, 2018-03-12:
@andreasmuelder don't know, haven't seen it in a long time. Closing for now, I think.

closed 2018-03-12 by BeckmaR