itemisCREATE/statecharts #1919: Integer comparison is not working in Yakindu Professional

byrzish, 2018-02-20:
With Yakindu Professional version 3.1.1 integer comparison is not working
in transition guards. If a variable is defined like this: var test: integer,
then in the transition if I try: [test == 1], I get error "Comparison operator '==' may only be applied on compatible types, not on integer and integer". The same with events with value. If the event is defined like this: interface: in event test: integer, then in the transition if I try: test [valueof(test) == 1], I get the same error. This is working on the Standard edition version 3.1.1.

andreasmuelder, 2018-02-21:
I can not reproduce this problem.
@byrzish Can you update to latest YAKINDU Professional 3.2 and check if the error still occures?

andreasmuelder, 2018-03-01:
Please reopen with steps to reproduce if this error still occures in the current version

closed 2018-03-01 by andreasmuelder