itemisCREATE/statecharts #1920: Statechart contains invisible transition to parent state

elgeekomm, 2018-02-21:
The attached statechart contains a hidden transition to its parent state. We struggled to identify the cause of the unexpected behavior, and eventually discovered there was a transition to the parent state visible in the properties view but not in the diagram.

We pared the statechart down to the simplest representation of the problem. Unfortunately, we don't know what steps created this issue.

A workaround is to expand the parent state inline, at which point the transition becomes visible.

 [hiddentransition.sct.zip](https://github.com/Yakindu/statecharts/files/1742524/hiddentransition.sct.zip)

BeckmaR, 2018-03-06:
Thank you for the statechart! It's always nice to have such well documented examples. Sadly I cannot reproduce this issue, when I extract a subdiagram with a transition to the parent state, the special exit syntax is used and the transition is visible in both diagrams. It looks like something went wrong when the state was refactored.

elgeekomm, 2018-03-06:
Thanks for looking into it, and sorry I couldn't send you a before and after that caused the problem. Just to clarify, while you're not able to recreate the circumstances that led to this issue, when you look at the attached statechart, do you see a transition that is not visible on the diagram that should be visible?

BeckmaR, 2018-03-07:
Yes, I do see it. The reason for its invisibility is just that it is not possible to display this transition, it basically goes to a wormhole - from the subdiagram-layer to the parent statechart layer. There is no way to display this properly because in both diagrams it's either lacking its source- or its target state. That's why normally these transitions are cut and a region exit is inserted.

As we currently do not have any steps to reproduce, I will close this issue. If you ever happen to experience this issue again, please comment, and especially if anyone finds a way to reproduce this. I personally could not do it, and looking at the refactoring code, I have no idea how that could ever happen.

BeckmaR, 2018-03-07:
Too quickly - @andreasmuelder found a way to reproduce this. If a region from a composite state containing a transition to said composite state is copied into a subdiagram, the error occurs, because the copy action does not contain the needed transition reconnection. This needs to be fixed.

elgeekomm, 2018-03-07:
Great! I'm glad this proved helpful. Thanks for the deep dive and quick response.

andreasmuelder, 2018-03-12:
scheduling this for the next milestone

PR #2196 (2018-09-03): Remove invalid outgoing transitions when pasting a region
Fixes #1920
closed 2018-09-03 by tkutz