itemisCREATE/statecharts #1925: Strange debugger behavior: active state not highligted, and event gets lost

RainerKlute, 2018-02-21:
Consider the following situation:
* The debugging session suspended at the **the_state**.
* The user raises two events, _e1_ and _e2_, in the simulation view.

[image]

Since we are in the event-driven execution scheme, we should expect that both events enter the event queue and are processed one after the other, and in the order the user raised the events.

Okay, so let's click on "resume" and see what happens:

[image]

Oops, what's that? The simulation is suspended, but **the_state** is neither highlighted as being the active state (and there's no other state that could be the active one), nor does it have a green border, indicating we are at a breakpoint.

We can see that the transition triggered by _e2_ has been taken as expected, since the count variable _c2_ has been incremented. We can also see that _e1_ is _no longer displayed as being raised_ in the simulation view. Not good! Does _e1_ nevertheless still linger somewhere and trigger the other transition and increment _c1_ when we resume the simulation?

[image]

No, it doesn't! Event _e1_ is lost. :-1:

BeckmaR, 2018-02-22:
The loss of event e1 is expected, because it is an `in event`. Only internal events are put into the queue.

RainerKlute, 2018-02-22:
Is expected? By whom? At least not by me. As a user, I cannot see where and why any _in_ event would ever get lost.

BeckmaR, 2018-02-22:
As I said before, an `in event` is not queued. Therefore the lower priority transition is never taken because after a run cycle in events are cleared. Raising two in events at once isn't possible with generated code (in the event driven case) because raising the event directly triggers a run cycle. That means that in real code you could only do this:

1. Raise event e1 (or e2, doesn't matter).
2. Raising the event triggers a run cycle, the left transition is taken and c1 incremented.
3. The raise event function returns
4. Raise event e2
5. Raising the event triggers a second run cycle, the right transition is taken, c2 is incremented.

for the regular CycleBased behavior, you could do this:

1. Raise event e1
2. Raise event e2
3. Trigger run cycle
4. The right transition is taken, c2 is incremented
5. Run cycle completes and clears all remaining in events.

terfloth, 2018-02-22:
@BeckmaR is right in respect to the execution of internal events during a run to completion step in event driven mode.  This is about correctly processing events in an internal event queue. 

@RainerKlute is right in respect to the overall behavior of the simulator from the end users point of view. This should also implement an external event queue where all 'in events' should go without any loss.

The difference between external and internal event queues is that the events in the external queue lead to separate run cycles and those in the internal queue are processed within a run cycle.

So we have different issues here:

1. Highlighting does not work properly.
2. We need an external event queue.

BeckmaR, 2018-02-28:
@RainerKlute is it possible that the state is not highlighted _yet_? What happens after a resume? A breakpoint in traditional source code means "stop before entering this line", so it would be reasonable not to highlight this state as active, as long as it becomes active after resume.

RainerKlute, 2018-02-28:
@BeckmaR, you are right that in traditional programming the debugger stops *before* executing the line in question. Nevertheless the debugger highlights exactly that line in the source code, so the programmer knows where the execution flow is and what comes next.

As a user, I'd expect the same when debugging statecharts, i.e., the debugger/simulator should clearly identify which of my breakpoints is the one that caused the simulation to suspend. In the example above, I have a single breakpoint only, and since it is attached to a state, that state should be highlighted as the current breakpoint.

When we speak about highlighting, perhaps it makes sense to clearly distinguish between
* highlighting an active state (yellow background) and
* highlighting a breakpoint state (green border).
So it might be sensible to draw a green line around a breakpoint state, while the previous state is still being the active one and is thus drawn with a yellow background. In the example, however, both breakpoint state and active state happen to be the same.

Anyway, in the example, **the_state** is _always_ active and must therefore _always_ have a yellow background.

BeckmaR, 2018-02-28:
Yeah if you look at it this way you are right. I believe in your screenshot the transition is highlighted. That doesn't make too much sense either because the simulation should have left it...

RainerKlute, 2018-02-28:
I don't understand your last sentence, @BeckmaR. Anyway, in screenshots 2 and 3 the state is neither highlighted as being active nor as being the current breakpoint state, while it should.

BeckmaR, 2018-02-28:
In your screenshots the transition is highlighted in orange as "active", which is not the correct behavior for a state breakpoint. It should stop after leaving the transition and before evaluating anything contained in the state.

RainerKlute, 2018-02-28:
Agreed

andreasmuelder, 2018-03-21:
The highlighting is resolved, moving this to S44

tkutz, 2018-09-26:
I need some more input for this issue.

I extended the example with the following:

[image]

Now, when I trigger both events manually, the corresponding transition is taken. It works for internal and in events. And it works for event driven and cycle based semantics.

So, in the simulator we have both events queued, but they are only visible within one run cycle.

For me, it kind of works as expected. 

@terfloth: Is there really any need to change the interpreter then?

andreasmuelder, 2018-11-06:
This will be resolved when external event queues are added to the simulator.

terfloth, 2018-11-20:
@tkutz the example reveals another aspect of the missing event queue and also another defect of the simulation engine. In event driven execution events must be processed one after the other. So no two events must be active at the same time.

terfloth, 2018-11-20:
Parts of this issue are already fixed, the remaining aspect that are caused by missing external event queue in the interpreter. For this part a new issue #2433 is created. Thus we close this issue as duplicate.

closed 2018-11-20 by terfloth