itemisCREATE/statecharts #1931: Conflict between Yakindu SCT installer and Yakindu Solidity IDE installer

RioVic, 2018-03-03:
It is not possible to install both SCT and Solidity IDE plugins in Eclipse due to library conflicts.   This makes it difficult for new Yakinduers who are working in the blockchain to get started and to appreciate the power of the toolkit to develop a DSL.

Workaround is to download both repos and manually sync the dependencies.   See comment below from Andreas Mulder:

The YAKINDU Solidity IDE reuses the expressions DSL from YAKINDU Statechart Tools, I think this is why you get such a bunch of errors.

To put it together I recommend to check out the latest master of Yakindu SCT and the Solidity IDE.

1. Download latest Eclipse Oxygen and install Xtext 2.13.
2. Clone our Statechart Tools Git repository from here:
https://github.com/Yakindu/statecharts
3 Import the following 4 projects in your workspace:
 - org.yakindu.base.expressions
 - org.yakindu.base.expressions.ui
 - org.yakindu.base.types
 - org.yakindu.base.xtext.utils.jface
4. Generate Expressions.xtext language
5. Clone the YAKINDU Solidity Repository
6. Import all Projects
7. Generate Solidity.xtext language
8. Open com.yakindu.solidity.target / Oxygen.target / resolve it and set as target platform.

After those steps you should have a workspace without compile errors.

andreasmuelder, 2018-03-12:
Thanks for reporting, this will be resolved with the next release of the YAKINDU Solidity IDE.

tkutz, 2018-09-03:
Fixed with YAKINDU Solidity 0.7.3

closed 2018-09-03 by tkutz