itemisCREATE/statecharts #1939: [Diagram Editor] Toggle documentation shortcut removes syntax highlighting

robert-rudi, 2018-03-14:
If the "Toggle documentation" shortcut in the toolbar is triggered to show documentation of i.e. states, an applied syntax highlighting is disabled for the whole statechart model.
I would expect, that some documentation is shown only for elements where a documentation is set, all other diagram elements should not change.

tkutz, 2018-07-10:
I can observe following issues:
* Documentation mode removes highlighting of expressions
* In pinned mode, definition section does not show documentation in documentation mode
* If no documentation sepcified, definition section is empty in documentation mode (in contrast to e.g. transitions which still show their expressions)

BeckmaR, 2018-07-10:
Another issue:

When the documentation and the actual code in a state differ a lot in size, toggling documentation mode toggles the state's size as well. Needless to say, this nukes the statechart's layout to dust.

tkutz, 2018-07-10:
With PR https://github.com/Yakindu/statecharts/pull/2130 the syntax highlighting problem is solved because in documentation mode, expressions are not shown in case of no documentation is provided.

The remaining problem is that toggling documentation does not work for the new pinned definition section.

closed 2018-07-11 by BeckmaR