itemisCREATE/statecharts #1947: Missing label in generated documentation

Bananeweizen, 2018-03-17:
When trying to open the documentation, I see this in the error log:
`Required attribute "label" missing from "topic" element in "/org.yakindu.sct.doc.user/help/user-guide/c-domain-toc.xml" (skipping element)
`

Indeed, c-domain-toc.xml contains this in line 3:
`<topic href="help/user-guide/c-domain.html" label=""></topic>`

svenjawendler, 2018-03-20:
This is due to h1{%EARLY_SEPARATOR%}. @RainerKlute I do not want to damage something. Can you take a look at this?

RainerKlute, 2018-06-29:
Yes, it is. It doesn't do any harm that I'd be aware of (except for the warning), so please bear with it.

closed 2018-06-29 by RainerKlute

paphko, 2020-08-31:
[image]

It's actually reported as errors when opened in a new (empty) workspace.