itemisCREATE/statecharts #1962: interpreter does not process internal events correctly in event driven execution mode

terfloth, 2018-03-20:
The current implementation of the interpreter does not process internal events correctly. It seems that an internal event is just partly processed for one region.

The attached project includes a test model and a unit test. The unit tests is successful with the last release but fails with a current product build.

We have to check what changed in the interpreter implementation and / or the model sequencer since the last release.

terfloth, 2018-03-20:
[event_driven_test.zip](https://github.com/Yakindu/statecharts/files/1828868/event_driven_test.zip)

andreasmuelder, 2018-03-20:
may be a regression introduced with https://github.com/Yakindu/statecharts/issues/1880

andreasmuelder, 2018-03-20:
This is also not working in release 3.2.0.

terfloth, 2018-03-20:
@andreasmuelder  your assumption was right - the cause was introduced with #1880 .
Debugging more deeply i came across another (more fundamental and ugly) problem.  The simulation engine executes some extra cycles. ...

I will take the ticket and go for a testing deep dive...

andreasmuelder, 2018-03-20:
the problem here is a duplicate enumeration, I fixed this and will create a PR.

PR #1969 (2018-03-21): Issue 1962
#resolves #1962
closed 2018-03-21 by terfloth