itemisCREATE/statecharts #1979: Exception while changing transition style

terfloth, 2018-03-21:
1. Select a transition
1. change transition style to 'Oblique' in properties view / appearance

The following exception is raised:

java.lang.ClassCastException: org.yakindu.sct.ui.editor.editparts.TransitionExpressionEditPart cannot be cast to org.eclipse.gmf.runtime.diagram.ui.editparts.ConnectionEditPart
	at org.eclipse.gmf.runtime.diagram.ui.properties.sections.appearance.ConnectionAppearancePropertySection.updateModel(ConnectionAppearancePropertySection.java:110)
	at org.eclipse.gmf.runtime.diagram.ui.properties.sections.appearance.ConnectionAppearancePropertySection.access$1(ConnectionAppearancePropertySection.java:97)
	at org.eclipse.gmf.runtime.diagram.ui.properties.sections.appearance.ConnectionAppearancePropertySection$4.widgetSelected(ConnectionAppearancePropertySection.java:252)
	at org.eclipse.swt.widgets.TypedListener.handleEvent(TypedListener.java:249)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:86)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4258)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1502)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1525)
	at org.eclipse.swt.widgets.Widget.sendEvent(Widget.java:1510)
	at org.eclipse.swt.widgets.Widget.notifyListeners(Widget.java:1314)
	at org.eclipse.swt.widgets.Display.runDeferredEvents(Display.java:4082)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3699)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1150)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1039)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:680)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:594)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:151)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
	at org.eclipse.equinox.launcher.Main.main(Main.java:1472)

andreasmuelder, 2018-03-21:
This can not be reproduced with the current master.
@terfloth please provide further steps to reproduce

robert-rudi, 2018-03-22:
For me, this exception occurs, if you try to switch the routing styles with the options "Avoid obstructions" or "Closest distance" activated.

commit de03e33e (2018-03-22): Fixes #1979. When switching routing options for transitions in the properties section, the input is now correctly processed. (#1994)
closed 2018-03-22 by andreasmuelder