itemisCREATE/statecharts #2041: NPE when changing the routing style and save / close state chart editor

wohaug, 2018-04-19:
Create a new state chart with two states and a transition and save it.
Then change the routing style of the transition and save / close the state chart.
NPE below occurs.

Observed in
YAKINDU Statechart Tools Standard Edition
version 3.3.1

`
java.lang.NullPointerException
	at org.eclipse.gmf.runtime.common.ui.services.action.global.GlobalActionHandlerService$ProviderDescriptor.provides(GlobalActionHandlerService.java:83)
	at org.eclipse.gmf.runtime.common.core.service.Service.safeProvides(Service.java:859)
	at org.eclipse.gmf.runtime.common.core.service.ExecutionStrategy$1.getUncachedProviders(ExecutionStrategy.java:88)
	at org.eclipse.gmf.runtime.common.core.service.Service.getProviders(Service.java:537)
	at org.eclipse.gmf.runtime.common.core.service.ExecutionStrategy$1.execute(ExecutionStrategy.java:67)
	at org.eclipse.gmf.runtime.common.core.service.Service.execute(Service.java:651)
	at org.eclipse.gmf.runtime.common.ui.services.action.global.GlobalActionHandlerService.execute(GlobalActionHandlerService.java:221)
	at org.eclipse.gmf.runtime.common.ui.services.action.global.GlobalActionHandlerService.getGlobalActionHandler(GlobalActionHandlerService.java:231)
	at org.eclipse.gmf.runtime.common.ui.action.global.GlobalAction.getGlobalActionHandlerData(GlobalAction.java:285)
	at org.eclipse.gmf.runtime.common.ui.action.global.GlobalAction.getGlobalActionHandlerData(GlobalAction.java:354)
	at org.eclipse.gmf.runtime.common.ui.action.global.GlobalAction.getGlobalActionHandlerData(GlobalAction.java:255)
	at org.eclipse.gmf.runtime.common.ui.action.global.GlobalAction.refresh(GlobalAction.java:147)
	at org.eclipse.gmf.runtime.common.ui.action.AbstractActionHandler.selectionChanged(AbstractActionHandler.java:387)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramGraphicalViewer.flushSelectionEvents(DiagramGraphicalViewer.java:256)
	at org.eclipse.gmf.runtime.diagram.ui.parts.DiagramGraphicalViewer$2.run(DiagramGraphicalViewer.java:269)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:37)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:182)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4213)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3820)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$5.run(PartRenderingEngine.java:1150)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.run(PartRenderingEngine.java:1039)
	at org.eclipse.e4.ui.internal.workbench.E4Workbench.createAndRunUI(E4Workbench.java:153)
	at org.eclipse.ui.internal.Workbench.lambda$3(Workbench.java:680)
	at org.eclipse.core.databinding.observable.Realm.runWithDefault(Realm.java:336)
	at org.eclipse.ui.internal.Workbench.createAndRunWorkbench(Workbench.java:594)
	at org.eclipse.ui.PlatformUI.createAndRunWorkbench(PlatformUI.java:148)
	at org.eclipse.ui.internal.ide.application.IDEApplication.start(IDEApplication.java:151)
	at org.eclipse.equinox.internal.app.EclipseAppHandle.run(EclipseAppHandle.java:196)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.runApplication(EclipseAppLauncher.java:134)
	at org.eclipse.core.runtime.internal.adaptor.EclipseAppLauncher.start(EclipseAppLauncher.java:104)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:388)
	at org.eclipse.core.runtime.adaptor.EclipseStarter.run(EclipseStarter.java:243)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
	at java.lang.reflect.Method.invoke(Unknown Source)
	at org.eclipse.equinox.launcher.Main.invokeFramework(Main.java:653)
	at org.eclipse.equinox.launcher.Main.basicRun(Main.java:590)
	at org.eclipse.equinox.launcher.Main.run(Main.java:1499)
`

robert-rudi, 2018-04-19:
Hello, thank you for this report.
We couldn't reproduce this issue with YAKINDU Statechart Tools v3.3.1

Today we will release v3.3.2.

If you can reproduce your issue with the new version, feel free to reopen this ticket.

wohaug, 2018-04-19:
When I reproduced this I installed a new and fresh version of YAKINDU SCT 3.3.1.
The NPE does NOT appear on ui but I can find it in the .log file of the workspace.
Will recheck once 3.3.2 is released.

robert-rudi, 2018-04-19:
You were absolutely right. 
I could now reproduce this issue with a fresh workspace and noticed the entry in the error log after closing the statechart.
This issue was already in v3.2.0. We will take this bug into our planning. 

Thanks again!

robert-rudi, 2018-05-23:
We are looking at this issue right now to provide a fix for our next release.
The source of this issue is in the underlying GMF GlobalAction infrastructure, so it was already present in version 2.6.0 of YAKINDU Statechart Tools.
For now, you can ignore the entry in the ErrorLog. It does not affect the statechart models you are working with.

robert-rudi, 2018-06-21:
This is a GMF related issues. I've submitted an eclipse bug report see https://bugs.eclipse.org/bugs/show_bug.cgi?id=536128.

I will close the issue until there are updates regarding the bug report.

closed 2018-06-21 by robert-rudi