itemisCREATE/statecharts #2045: always, oncycle, every and after may be used as names for 'in events'

herrendorf, 2018-04-21:
The internal event names `always`, `after`, `oncycle` and `every` may be used as names for `in events`:

```
@CycleBased(200)

interface:
// Define events and
// and variables here. 
//Use CTRL + Space for content assist.

in event always 
```

This causes an ambiguity, if the event is used on  transition.

herrendorf, 2018-04-21:
`after` and `every` already break the parser and cause a `STextParser$SyntaxEception`. So an error is shown, but the message is rather confusing:
```
No viable alternative at input 'after'
Qualified name cannot be null```

andreasmuelder, 2018-05-02:
same for exit and entry triggers

andreasmuelder, 2018-05-02:
fixed by https://github.com/Yakindu/statecharts/pull/2057

closed 2018-05-17 by BeckmaR