itemisCREATE/statecharts #2066: < and > signs being replaced with 0 and 2 in C# code generation

stuartdl, 2018-05-09:
I have built the C# code generator from the master branch and have tried to compile the generated C# code.
The issue found In the resulting code is, when a comparison between two values is meant to be made, using the '<' and the '>' symbols, the symbols are replaced with a 0 (zero) and a 2 
for example in the function below
`
private bool check_CSharpErrorDemo_main_region_ValveClosed_lr0_lr0()
{
return (doSomething
) && (actualPressure0targetPressure
);`

The last line SHOULD be 

`) && (actualPressure < targetPressure`

I have attached a project that demonstrates this issue and the code above was taken from the generated file called "CSharpErrorDemoStatemachine.s" on line 97

[CSharpErrorDemo.zip](https://github.com/Yakindu/statecharts/files/1986269/CSharpErrorDemo.zip)

BeckmaR, 2018-05-15:
Hey, thank you for the report. Maybe you could try if the branch `issue_2066` fixes this? The C# generator doesn't get much love at the time being and I have no C# development environment to test it out myself, but in theory this should do.

stuartdl, 2018-05-16:
Hello Rene,
 Thankyou for that. I have downloaded it and tried to rebuild it but am
experiencing some build problems with my eclipse install (i'm very new to
eclipse). I did try an "ommph" install which worked perfectly last time but
this time there are a lot of errors and the  "Expressions", "SText", and
"SGen" languages are failing to build
Hopefully I can get it back up and running in the next couple so i can
build and try the plugin
Regards
Stuart

BeckmaR, 2018-05-16:
Oh, yeah, I experienced that issue as well. There was a bug concerning build paths on the master branch. I merged the fix into the branch issue_2066, so you should be able to build it now after a `git pull`.

stuartdl, 2018-05-21:
Hello Rene,
   Any chance you could email a copy of that pluggin to me?
I have tried a couple of times to build the code from scratch and although
I did have it building correctly a couple of weeks back, since I have done
the git pull, I have started to get many erorrs when trying to build.
Thanks
Regards
Stuart

stuartdl, 2018-05-22:
Hello Rene,
    Good news, I managed to get sct to build and have tested the C# plugin and that patch has fixed the issue.
Thanks for your help
Regards
Stuart
P.S. If there is any more development being done on the C# code generator, I'd be happy to test it :)

closed 2018-05-22 by stuartdl

BeckmaR, 2018-05-23:
Hey stuart, I'm sorry I couldn't get back to you earlier, but I was not in office until today. I'm glad to hear you managed to build sct yourself again, but of course it would not have been a problem to send you the plugin. If you encounter any more issues, please feel free to report them!

Regards,
René

PR #2082 (2018-05-24): Issue 2066
Fix the CSharp generator which had a weird bug with `<` and `>` operators.
Fix #2066