itemisCREATE/statecharts #2125: Event does not drive event-driven statechart

RainerKlute, 2018-07-06:
The following statechart should continuously execute the dicing loop. However, it gets stuck in the *Rolling_the_dice* state. It seems that the _ev_ event raised in the state has either not been raised at all or is not processed correctly. In cycle-drive mode, the behavior is the same.

[image]

BeckmaR, 2018-07-18:
I can confirm this issue - weird, I thought this works.

BeckmaR, 2018-07-18:
This works when this is not the first state. The reason is that Entering the state machine does not trigger a full run cycle. The expressions in the initial state are evaluated though, but the internal event queue is not checked after that. Raising any event on the statemachine would make it run.

However, this particular statechart would of course run into an infinite loop, which is probably not desired. A stop condition should be considered.

BeckmaR, 2018-07-31:
After inspecting the generated code it becomes clear that this is indeed the expected behavior. The event gets raised as expected, but "entering" the statechart is not doing a runCycle, _which is fully consistent with all code generators_. A workaround would be to either just raise some event on the statechart or instead of entry declare a time event, which will trigger a runcycle when it elapses.

closed 2018-07-31 by BeckmaR